terraref/reference-data #87: Review structure of genomic data at CyVerse

nfahlgren, 2017-02-07:
@dlebauer and @craig-willis could you review the structure of data in our CyVerse data repository?

All TERRA-REF raw genomic data has been uploaded to CyVerse here `/iplant/home/shared/terraref`. 

The data is structured as follows:

```
/terraref
    /sites
        /hudson-alpha
            /raw_data
                /year1
                /year2
            /derived_data
```

The main folder contains README.txt and DATA_USE_POLICY.md files. I added a table of sample information to the year1 and year2 subfolders, based on https://docs.google.com/spreadsheets/d/1ntnps0u0c8O4xxrv_VVqfxDkpmppCGdFIIWLMmRCeSM/edit?usp=sharing.

Questions:

- I am sure the README could be better, but anything missing that's needed right now?
- Should the data use policy document be duplicated in any subdirectories?
- Are the sample tables in the year1 and year2 subdirectories okay for now?

Any other suggestions welcome.

craig-willis, 2017-02-09:
@nfahlgren Should I be able to see this in the DataCommons without logging in ?

nfahlgren, 2017-02-10:
@craig-willis no, it's not public yet. Can you see it when you are logged in?

craig-willis, 2017-02-10:
@nfahlgren I'm not sure where to look.  I don't see it in http://datacommons.cyverse.org/, nor in https://de.cyverse.org/ under "Community data" or "Shared with Me." I'm new to Cyverse and probably looking in the wrong place.

nfahlgren, 2017-02-10:
@craig-willis I think it won't show up in Data Commons until we make it public, but it is in the Community Data space, so that's where it will show up at some point. I think the easiest way to get to it now is through the Discovery Environment, and I was able to generate a link to the folder: https://de.cyverse.org/de/?type=data&folder=/iplant/home/shared/terraref. I can't or don't know how to change the permissions on the folder, so if the link doesn't work @dlebauer may need to add you to the access list.

phloz, 2017-02-10:
@craig-willis @nfahlgren 
I can see the data in the Cyverse Discovery Environment Community Data folder in the follow path /iplant/home/shared/terraref/sites/hudson-alpha/raw_data/ as Noah referenced above

craig-willis, 2017-02-10:
Thanks, @nfahlgren. I don't seem to have access to the "terraref" folder and will check with @dlebauer

craig-willis, 2017-02-23:
@dlebauer Can I get access to this folder in Cyverse?

dlebauer, 2017-02-23:
Do you have a cyverse username?

craig-willis, 2017-02-23:
willis8

dlebauer, 2017-02-23:
Noah has the permissions - Noah can you add Craig and I?

nfahlgren, 2017-02-23:
@dlebauer I can't figure out how to do this on cyverse, how did you do it before?

nfahlgren, 2017-02-23:
@craig-willis I think you should have access to `/iplant/home/shared/terraref/` now

craig-willis, 2017-02-23:
@nfahlgren Indeed I do. Thank you.

dlebauer, 2017-03-02:
looks great. go ahead and make it public

closed 2017-03-02 by dlebauer