terraref/reference-data #88: Check how plantcv metadata is stored

dlebauer, 2017-02-11:
In the sample datasets on Clowder the experiment metadata is at the dataset level

[image]

 and genotype is found at the file level way after the exif info 

[image]

Seems like the genotype + barcode + treatment etc should be at the dataset level. 

Would that make sense?

nfahlgren, 2017-02-11:
@dlebauer or both? But yeah, I agree that those metadata apply down from the dataset level.

It would also be useful to look over MIAPPE and see how it compares to what we have.

dlebauer, 2017-02-11:
Both works. And perhaps relevant metadata at the collection level too?

MIAPPE is defined here http://cropnet.pl/phenotypes/wp-content/uploads/2016/04/MIAPPE.pdf

Looks like you are the person to provide this information Noah  but @craig-willis may be able to advise on how much we can cover with linked data vs duplication; seems a search should return miappe info + data

nfahlgren, 2017-02-11:
I need to rework the PlantCV code in the current extractor to analyze the TERRA-REF datasets, based on scripts I got from Todd's group. That would be a good time to incorporate updates to the metadata

rachelshekar, 2017-05-18:
@max-zilla will look into limitations to EXIF

nfahlgren, 2017-05-18:
Database extractor: https://github.com/danforthcenter/pheno-data-service

rachelshekar, 2017-05-18:
Noah wants users to enter experimental data into database and extract that information into EXIF during downstream analysis

dlebauer, 2017-06-08:
@nfahlgren see https://github.com/terraref/reference-data/issues/137 and contact me if you need assistance.

closed 2020-06-05 by dlebauer