terraref/reference-data #96: Soil removal and compensation for shadow effects

remotesensinglab, 2017-03-02:
### Description

Removal of soil component and compensation or correction for shadow effects need to be incorporated in hyperspectral calibration. These effects may affect our plot level summaries. Shadow may not necessarily noise. It is possible that leaves under the shadow of another plant so it shows a 'plant-like' spectra. It is possible to compensate for shadow rather than removal. 

### Completion Criteria   

* [ ] Discuss and develop requirements docs; create issues for next steps
for a new feature
* [ ] create algorithm
* [ ] write test
* [ ] add to pipeline

dlebauer, 2017-03-02:
Can we start by creating masks for soil and shade, each a binary layer (0,1) at Level 1 resolution, indicating whether the pixel contains soil or shade? These could be level 2 products

Paheding, 2017-04-06:
@dlebauer Yes, in current stage, we can create binary masks for soil and shade. Meanwhile, we are testing our shadow compensation technique extensively on various data sets. Once it is mature enough, we can deploy it.  Thanks.

dlebauer, 2017-04-06:
@Paheding do you have an estimated timeline for this? Do you think we could have something by the end of April? It would be great if we could deploy this alongside the spectral index products (late April). 

I'd like to re-emphasize that while it should be reasonably accurate, we are using an iterative deployment process, and each year (early fall) we will re run any algorithms that have been updated.

rachelshekar, 2017-04-06:
@Paheding Please talk to Erika to determine if the shadow work can be included in the 2017 or 2018 release.

rachelshekar, 2017-05-17:
@Paheding - can you give us an update on this?

remotesensinglab, 2017-05-17:
Hi David,
Soil and shadow removal algorithms are already implemented in the workflow, I believe. Shadow compensation part is not. I think we need to evaluate that part with the new diurnal data collected. 

Wasit

Paheding, 2017-05-18:
@rachelshekar We can release the shadow work into the private repository for this year.  Actually, we have already deployed the soil and shaded soil removal algorithms into the private repository and it's been running as an extractor.

dlebauer, 2017-05-18:
@Paheding Can you point to an example of the output generated by the extractor on ROGER? 

Did you write tests for the extractor?

Paheding, 2017-05-18:
@dlebauer   The extractor generates a mask for soil/shaded soil and saves the mask as a .npy  file. Here is an example output (White: plant; Black: Soil):

[image]

I believe @max-zilla converted the code into Clowder script format and sent it to run.

dlebauer, 2017-05-18:
@ZongyangLi please export as raster netcdf file (rather than or in addition to .npy file) - either as a new layer in the input file or as a separate file.

ZongyangLi, 2017-05-18:
@dlebauer I am not quite clear about this but I guess @Paheding is working on this?

Paheding, 2017-05-18:
@ZongyangLi  Currently, the output of the mask is a .npy file. @dlebauer suggests exporting it to a necdf file as a new layer in the input file. I think @max-zilla can point the output location of the mask.

max-zilla, 2017-05-19:
@ZongyangLi I think you were incorrectly tagged.

@dlebauer @Paheding currently the soil_removal extractor generates an .npy file which is just a numpy array, I included methods in terrautils for conversion of that to geoTIFF or PNG. this seems like a good excuse to add a netCDF method as well.

I believe @czender has already done much of this work for us in the hyperspectral DataProcess.py script:
https://github.com/terraref/extractors-hyperspectral/blob/master/hyperspectral/DataProcess.py#L103

pseudo:
```
newData = Dataset("newfile.nc","w",format="NETCDF4")
dimensionX, dimensionY = getDimensions(numpyArray)
newData.createDimension('band',  1) # only 1 band for mask 
newData.createDimension('x', dimensionX)
newData.createDimension('y', dimensionY)
mask = newData.createVariable('soil_mask','f8',('band', 'x', 'y') # not sure about this but just needs 1/0
mask[:] = numpyArray
newData.close()
```

Paheding, 2017-05-19:
@max-zilla Thanks for converting that. The geoTIFF version would be more helpful compared to .npy.

Paheding, 2017-05-22:
@dlebauer @max-zilla I have modified the extractor (**terra_soil_removal.py**) to save the soil removal mask as a netCDF file.

closed 2017-06-01 by rachelshekar