terraref/reference-data #100: Link genotypes with sites

dlebauer, 2017-03-06:
Many traits in bety have a plot name but not a genotype name. These need to be updated.

For Sorghum Season 1: link cultivars and sites tables from this spreadsheet on the field 'entry' https://docs.google.com/spreadsheets/d/1s-kQc3K0h4-hfwMzKOU_zpRXN5loByIpKW8Rvw76mh0/edit#gid=796817704

then each row in traits table should have a cultivar_id that matches the 'new_cultivar_name' field. The match with site name can be partial (since some site names have 'E' or 'W' appended.

gsrohde, 2017-06-28:
@dlebauer I have written a script to make these links.  But many of the cultivar names are not found.  I'm guessing this is only because there has been some revising of names.  For example, although there is no "PI_35038", there is "PI35038".

If I am correct that it was decided to do away with underscores in the names starting with "PI", then, rather than rewriting the CSV file, I can just hack the script to change every prefix of the form "PI_" to just "PI".  Then everything is found.  Shall I do this?

dlebauer, 2017-06-28:
Yes please do. We decided to drop the "_"

gsrohde, 2017-06-29:
Done.  If you think we'll do this sort of thing periodically, I'll polish up the script a bit and commit it to the Rails repository.

rachelshekar, 2017-07-12:
@dlebauer can this be closed?

closed 2017-08-24 by rachelshekar

dlebauer, 2017-08-24:
@gsrohde please post the script as a gist and link to it from this issue. If we need to do this again, we can revisit whether to clean up / generalize it.