terraref/reference-data #98: Incorrect date (year 1016) in raw_data

dlebauer, 2017-03-06:
We have a year 1016 that should be 2016. This appears in the raw data folder on Roger/Globus

sites/ua-mac/raw_data/scanner3DTop/1016-12-09/

and this has been propagated into Clowder, e.g. https://terraref.ncsa.illinois.edu/clowder/collection/587126114f0cc129fb5740d7

### Completion Criteria   

* [ ] Identify the source of this error ( @smarshall-bmr )
* [ ] Fix error ( @jdmaloney )

jdmaloney, 2017-03-06:
Fixed the directory name on ROGER, new check list item now.  

- [ ] Update in Clowder @max-zilla 

We're doing some high-tech stuff here in 1016 ;)

max-zilla, 2017-03-06:
@jdmaloney ah yeah, I had corrected this in the Clowder dataset name but not on the filesystem. I'll update the affected dataset pointer and close this

rachelshekar, 2017-04-20:
do we still need @smarshall-bmr  to find the source of the problem?

smarshall-bmr, 2017-04-24:
@dlebauer I don't actually have access to the program that writes dates to the filenames, but it seems there was a simple typo. I'm checking with LemnaTec now about how far this would propagate, but is this just on the 3D data or is it on all the data from that date on our amazing time traveling field scanalyzer?

dlebauer, 2017-04-24:
@smarshall-bmr I don't see any more directories matching `sites/ua-mac/raw_data/*/1*`. 

@jdmaloney and @max-zilla can this be closed?

closed 2017-05-18 by max-zilla