terraref/reference-data #99: Reupload canopy_height and spike_height with dates for Season 1

dlebauer, 2017-03-06:


gsrohde, 2017-04-03:
@dlebauer It seems this data has _already_ been uploaded _twice_, once with dates and once without.  The first time it was uploaded was last August 12—in fact, these are the first traits in the database, and dates are included.  The second time, without dates, was November 10.

I'm not exactly sure how this happened, but my best guess is that I was provided essentially the same data twice under differently-name files and in different issues.  The November 10 upload was part of issue https://github.com/terraref/reference-data/issues/61.  The earlier upload was possibly in connection with issue https://github.com/terraref/reference-data/issues/46 (definitely not issue 61 since that dates from October) or maybe no issue at all.  All I can say for sure is that the file names I used for the upload data were ht1.csv (actually, ht1-amended.csv) and ht2.csv, but these may be names I made up to abbreviate the impossibly-long names that Google Docs provides.

I have expressed more than once, I think, my concern about the possibility of the same data being uploaded twice.  Once this happens, it becomes a major forensic investigation to figure out what has been done and how to fix it, time I could be spending making software improvements.  I think it likely that there are other cases where the same data has been uploaded twice.  Since last August, I have tried to keep increasingly careful track of what files I upload, what modifications I had to make to the provided files, what the id numbers of the resulting new rows were, etc., but my record-keeping has been less than perfect, and even with a record of what I have done, it isn't always easy to avoid or correct mistakes.  This is all complicated by the fact that issue comments and Google Doc files (and possibly their names) are sometimes revised after I do an upload.

The file you provide above, by the way, doesn't provide any new information not already in the file provided in issue 61 as far as I can tell, and moreover, the site names are incorrect.  So I hope you didn't spend too much time creating it.  (If there is something I am missing, let me know.)  The only reason the upload I did in connection with issue 61 didn't insert dates is that I used the API for the upload, which requires the date column heading to be "local_datetime" rather than "date".  Possibly it should be an error if unrecognized column headings are used in API-upload files.  It should at least be an error if there is a column heading "date" for API upload files.  I will make an issue for this.  The Bulk Upload Wizard at least warns when a column will be ignored.

dlebauer, 2017-04-20:
please

* go ahead and leave the records the ones with dates, delete the ones without, and close issue; 
* open an issue for preventing duplicates (esp via constraints)

gsrohde, 2017-06-14:
Most of this data was first uploaded on August 12, 2016.

It was uploaded again on November 10, 2016.

The names of the file used for the August 12 upload were `ht1-amended.csv` and `ht2.csv`.  `ht1-amended.csv` contained only canopy_height data; `ht2.csv` contained both canopy_height and spike_height data.  The data in these files were uploaded using the API and the Bulk Upload Wizard, respectively.  (The Bulk Upload Wizard was probably used for `ht2.csv` because it has two traits per row, and the API did not at the time fully support this.)

The November 10 upload was done with the API.  Again, two files were used, one containing only canopy_height data and the other containing both canopy_height and spike_height data.  **_The latter file contained 27 rows of data not in the `ht2.csv` file used for the August 12 upload._**  The date column in each file was ignored because it was incorrectly named.  In addition, the CSV files used for the November 10 upload contained no treatment or cultivar information, so the resulting rows are missing values for treatment_id and cultivar_id.

# The id number correspondences are as follows:

The rows with ids 6000000001–6000000200 correspond to the rows with ids 6000021253–6000021452.  These were the rows resulting from the data files that contained both canopy_height and spike_height data.  (As mentioned above, the CSV file for the November 10 upload contained 27 rows not in the file for the earlier upload, and these rows correspond to the database rows with ids 6000021453–6000021506.)  That these sets of rows correspond is shown by the query

SELECT
*
FROM
traits t1
CROSS JOIN traits t2
WHERE
t1. ID = t2. ID - 21252
AND t1. ID BETWEEN 6000000001 AND 6000000200
AND t2. ID BETWEEN 6000021253 AND 6000021452
AND (
t1.site_id != t2.site_id
OR t1.site_id IS NULL
OR t2.site_id IS NULL
OR t1.specie_id != t2.specie_id
OR t1.specie_id IS NULL
OR t2.specie_id IS NULL
OR t1.citation_id != t2.citation_id
OR t1.citation_id IS NULL
OR t2.citation_id IS NULL
OR t1.cultivar_id IS NULL
OR t2.cultivar_id IS NOT NULL
OR t1.treatment_id IS NULL
OR t2.treatment_id IS NOT NULL
OR t1. DATE IS NULL
OR t2. DATE IS NOT NULL
OR t1.dateloc != 5
OR t1.dateloc IS NULL
OR t2.dateloc != 9
OR t2.dateloc IS NULL
OR t1.mean != t2.mean
OR t1.mean IS NULL
OR t2.mean IS NULL
OR t1.variable_id != t2.variable_id
OR t1.variable_id IS NULL
OR t2.variable_id IS NULL
OR t1.entity_id IS NULL
OR t2.entity_id IS NULL
)
ORDER BY
t1. ID;

This returned 0 rows, showing that the sets correspond as regards site_id, specie_id, citation_id, mean, and variable_id (these are always non-null in both sets). (The values for timeloc, user_id, checked, and access_level also always happen to be the same (they are uniform). 

The differences are that the rows in the higher numbered set (corresponding to the November 10 upload) always have NULL cultivar_id, date, and treatment_id, and the dateloc value is always 9 rather than 5.  The entity_id values are different, but this is not significant because they are anonymous in both cases, and in both cases they correctly group canopy height and spike height measurements occurring in the same row of the input CSV file.  And of course the time stamp and id values are different.

The values in the other columns are either always null or always blank in both cases.

## Proposed fix for eliminating duplicates:

* Delete rows 6000021253–6000021452.
* Add date, treatment, and cultivar information to rows 6000021453–6000021506 and change the dateloc value to 5.  These are rows from the November 10 upload that don’t correspond to any rows of the August 12 upload.

The SQL statements to effect this are

```
DELETE FROM traits WHERE id BETWEEN 6000021253 AND 6000021452;
UPDATE traits SET date = '2016-06-23 07:00:00’, dateloc = 5, treatment_id = 6000000014, cultivar_id = NULL;
```
[... 41 lines cut ...]
or t1.variable_id is NULL
or t2.variable_id is null
or t1.entity_id is NULL
or t2.entity_id is not null
) 
ORDER BY
t1. ID;
```

As with the previous matching sets, this returned 0 rows, showing that the sets correspond as regards site_id, specie_id, citation_id, mean, and variable_id.  Unlike the previous sets, however, the high-numbered set in this case always has NULL entity_id.  (This is because the API was used, and, unlike the Bulk Upload Wizard, it does not create anonymous entities when there is only one trait per row of the CSV file.)

## Proposed fix for eliminating duplicates:

* Delete rows 6000021507–6000028399.
* Set the entity_id to NULL for rows 6000000201–6000007093 and delete the entities to which they refer.

The reason to delete these entities is that they are each unnamed and associated with only a single trait and so they do not add any information.

The SQL statements to effect this are

```
DELETE FROM traits WHERE id BETWEEN 6000021507 AND 6000028399;
DELETE FROM entities WHERE id IN (SELECT entity_id FROM traits WHERE id BETWEEN 6000000201 AND 6000007093);
UPDATE traits SET entity_id = NULL WHERE id BETWEEN 6000000201 AND 6000007093;
```

gsrohde, 2017-06-14:
@dlebauer This turns out to be slightly more complicated than I at first thought.  I'll go ahead and do the fixes, but I wanted you to at least have a chance to look through and check my logic (and maybe my SQL statements) first.  If you are short on time, perhaps you can at least look through the two "Proposed fix for eliminating duplicates" sections.

dlebauer, 2017-06-14:
@gsrohde this looks correct for eliminating these duplicates. Would be useful to also have a constraint that prevents inserting duplicate records - either by the uniqueness constraint that includes the mean value.

gsrohde, 2017-06-14:
The deletions and updates are done.  I'll close this as soon as I make an issue for preventing duplicates.

closed 2017-06-15 by gsrohde