terraref/reference-data #101: Public data hidden in Clowder unles superadmin is enabled

dlebauer, 2017-03-07:
For many of the datasets and collections, I can see some data as SuperAdmin that is not normally visible, even when the files are marked public.

For example, https://terraref.ncsa.illinois.edu/clowder/datasets/5873a8ce4f0cad7d8131ad86

Without SuperAdmin:

[image]

With SuperAdmin

[image]

Is this a bug in Clowder or an issue with incorrect permissions? Can this be fixed globally?

### Completion Criteria

* [ ] make sure all publicly available data sets are visible in Clowder

max-zilla, 2017-03-07:
I will look into this - tagging @robkooper as well in case he spots something obvious.

robkooper, 2017-03-08:
This is a bug, if you are not logged in you will see the data. Seems when getting the data the logged in and not having access to the data overrides the public data flag.

Created clowder issue: https://opensource.ncsa.illinois.edu/jira/browse/CATS-744

max-zilla, 2017-05-31:
https://opensource.ncsa.illinois.edu/bitbucket/projects/CATS/repos/clowder/pull-requests/1141/overview I believe this pull request may solve this issue. The error I get on that dataset when logged in is the same (htmlEncode) and as superAdmin there's no such error.

closed 2018-05-31 by sbrad77