terraref/reference-data #103: Unclear what 'reflectance' data in calibration object metadata refers to

dlebauer, 2017-03-08:
In this file

https://github.com/terraref/sensor-metadata/blob/master/targets/LabSphere/fixed_metadata.json

It is not clear what the (the dimensions, units, etc) of the reflectance values are:

e.g.

```json
{
  "manufacturer": "LabSphere Inc.",
  "product_name": "Spectralon Diffuse Color Standards",
  "targets": {
	"black" : {
		 "model" : "SCS-BL-020",
		 "serial_number" : "BLAA01-0815-2506",
		 "reflectance" : [[38.9],[39.1],[39.3],[39.5],[39.7],[40],[40.2],[40.5],[40.7],[40.8],[41.2],[41.5],[41.7],[42],[42.3],[42.5],[42.8],[43.1],[43.5],[43.8],[44.1],[44.4],[44.7],[45],[45.3],[45.6],[46],[46.3],[46.6],[47],[47.3],[47.6],[48],[48.3],[48.7],[49],[49.4],[49.6],[49.9],[50.1],[50.3],[50.6],[50.8],[51],[51],[51.2],[51.3],[51.4],[51.7],[52.2],[52.9],[53.6],[54.3],[54.9],[55.4],[55.9],[56.4],[56.7],[57.1],[57.5],[58],[58.2],[58.4],[58.7],[59],[59.3],[59.6],[59.8],[60],[60.2],[60.5],[60.7],[60.9],[61.1],[61.2],[61.4],[61.4],[61.6],[61.7],[61.7],[61.8],[61.8],[61.9],[61.9],[61.9],[61.8],[61.7],[61.8],[61.7],[61.6],[61.5],[61.5],[61.4],[61.4],[61.4],[61.4],[61.5],[61.5],[61.4],[61.6],[61.6],[61.5],[61.4],[61.3],[61.1],[60.9],[60.8],[60.5],[60.3],[59.9],[59.6],[59.2],[58.9],[58.3],[57.6],[56.9],[56.1],[55.3],[54.8],[54.3],[54.2],[54.5],[55],[55.8],[56.6],[57.5],[58.1],[58.7],[59.3],[59.6],[60],[60.1],[60],[59.9],[59.7],[59.4],[59],[58.4],[57.6],[56.9],[56],[55.1],[54.2],[53.1],[52],[50.9],[49.8],[48.7],[47.6],[46.4],[45.2],[44],[42.9],[41.7],[40.5],[39.2],[38.1],[36.9],[35.7],[34.6],[33.5],[32.5],[31.4],[30.4],[29.5],[28.6],[27.7],[26.9],[26.1],[25.3],[24.5],[23.7],[22.9],[22.2],[21.5],[20.8],[20.1],[19.4],[18.8],[18.2],[17.7],[17.2],[16.8],[16.5],[16.2],[16],[15.8],[15.7],[15.6],[15.5],[15.4],[15.4],[15.5],[15.6],[15.6],[15.5],[15.2],[14.9],[14.7],[14.4],[14.1],[13.9],[13.7],[13.5],[13.2],[13],[12.9],[12.6],[12.4],[12.2],[12.1],[11.9],[11.7],[11.6],[11.4],[11.3],[11.2],[11.1],[11.1],[11],[10.9],[10.9],[10.8],[10.8],[10.8],[10.8],[10.8],[10.8],[10.8],[10.8],[10.8],[10.8],[10.9],[10.9],[10.9],[10.9],[11],[11],[11],[11.1],[11.1],[11.2],[11.3],[11.3],[11.4],[11.5],[11.6],[11.7],[11.8],[11.8],[11.9],[11.9],[11.9],[11.9],[11.9],[11.9],[11.8],[11.7],[11.7],[11.6],[11.6],[11.5],[11.5],[11.5],[11.5],[11.5],[11.6],[11.6],[11.7],[11.8],[11.9],[12.1],[12.2],[12.4],[12.5],[12.7],[13],[13.2],[13.5],[13.8],[14.2],[14.6],[15],[15.5],[16.1],[16.7],[17.3],[18],[18.7],[19.5],[20.3],[21.2],[22],[23],[24],[24.9],[26],[27.1],[28.2],[29.5],[30.7],[31.9],[33.2],[34.5],[35.8],[37.2],[38.6],[39.9],[41.3],[42.7],[44.2],[45.6],[47],[48.5],[49.8],[51.2],[52.6],[54],[55.4],[56.7],[58],[59.4],[60.6],[61.9],[63.1],[64.5],[65.6],[66.8],[68],[69.1],[70.2],[71.3],[72.3],[73.3],[74.2],[75.1],[75.9],[76.7],[77.4],[78.2],[78.8],[79.6],[80.1],[80.6],[81.1],[81.5],[82.1],[82.5],[82.9],[83.3],[83.7],[84],[84.3],[84.6],[84.9],[85.1],[85.3],[85.6],[85.7],[85.8],[86.2],[86.3],[86.5],[86.6],[86.7],[86.9],[87],[87.2],[87.3],[87.5],[87.6],[87.7],[87.8],[87.9],[87.9],[88],[88.1],[88.3],[88.3],[88.4],[88.5],[88.6],[88.7],[88.7],[88.9],[88.8],[88.9],[89],[89],[89.1],[89.1],[89.2],[89.3],[89.4],[89.3],[89.5],[89.5],[89.5],[89.6],[89.7],[89.7],[89.7],[89.8],[89.8],[89.8],[89.9],[89.9],[90],[90],[90],[90.1],[90.2],[90.1],[90.2],[90.3],[90.3],[90.2],[90.3],[90.4],[90.4],[90.4],[90.5],[90.4],[90.6],[90.7],[90.7],[90.6],[90.5],[90.6],[90.6],[90.6],[90.6],[90.7],[90.6],[90.6],[90.6],[90.7],[90.7],[90.7],[90.7],[90.8],[90.7],[90.7],[90.7],[90.7],[90.9],[90.8],[90.9],[90.8],[90.9],[90.9],[90.9],[90.9],[90.9],[90.9],[90.9],[90.8],[91],[90.8],[91],[90.7],[90.8],[90.9],[90.8],[90.9],[91],[91],[90.9],[90.7],[90.6],[90.9]]
	 },

```

### Completion Criteria   

* [ ] Make sure that these numbers have sufficient meta-data to interpret
* [ ] Review other sensor metadata in github.com/terraref/sensor-metadata to ensure the same

rachelshekar, 2017-06-22:
@dlebauer please assign this issue

craig-willis, 2017-06-22:
@dlebauer 

I'll need guidance on what's needed here.  The source data is from:
https://drive.google.com/drive/folders/0BzWFmpKimq_6T3BINnFIQW5uWlk

The table in the PDF has the heading:
"8 deg/Hemispherical Spectral Reflectance Factor for SCS-BL-020 Color Standard"

I'm out of my depth with the reflectance and calibration metadata and have a hard time judging what's needed.

dlebauer, 2017-06-22:
@solmazhajmohammadi and @smarshall-bmr could you please provide more information about the target metadata?

dlebauer, 2017-06-22:
It looks like %reflectance. The confusion is what wavelength bands the reflectances correspond to.

solmazhajmohammadi, 2017-06-22:
The corresponding wavelength is written before the reflectance. Is this the question?

Wavelength	Corrected %R
360	                   38.9
361	                   39.1
362	                   39.3
363	                   39.5
364	                   39.7
365	                   40.0
366	                   40.2
.....

dlebauer, 2017-06-22:
@solmazhajmohammadi I am referring to https://github.com/terraref/sensor-metadata/blob/master/targets/LabSphere/fixed_metadata.json ? 

Are the wavelengths all just sequential integers starting at 360?

solmazhajmohammadi, 2017-06-22:
@dlebauer yes, that's the table we got from Labsphere, and it matches the calibration certificate.

rachelshekar, 2017-09-07:
@dlebauer can this be closed?

craig-willis, 2018-06-13:
This is a stale issue with no comments in last 2 months. Please create new issues if work remains or put this content in the documentation.

closed 2018-06-13 by craig-willis