terraref/reference-data #104: Dublin Core metadata for the TERRA-REF Community Data folder at CyVerse

nfahlgren, 2017-03-09:
@dlebauer @craig-willis I am working with CyVerse to make the genomic data available through the CyVerse Data Commons. I was asked to add metadata to the `terraref` folder using the Dublin Core template. Could you please provide feedback on what I have below? Here are the metadata fields for DC and my proposed values in quotes:

**Title** (required): The name given to the resource. Typically, a Title will be a name by which the resource is formally known.

> Transportation Energy Resources from Renewable Agriculture Phenotyping Reference Platform (TERRA-REF) Genomic Data Repository for Sorghum

**Subject** (required): The topic of the content of the resource. Typically, a Subject will be expressed as keywords or key phrases or classification codes that describe the topic of the resource. Recommended best practice is to select a value from a controlled vocabulary or formal classification scheme.

> I am not sure what vocabulary we would use for subject terms

**Description** (required): An account of the content of the resource. Description may include but is not limited to: an abstract, table of contents, reference to a graphical representation of content or a free-text account of the content.

> The terraref directory contains raw and derived sorghum genome sequencing data from the Transportation Energy Resources from Renewable Agriculture Phenotyping Reference Platform (TERRA-REF) project (http://terraref.org/). Raw data includes DNA sequence files in compressed FASTQ format. Derived data was generated using CoGe (https://genomevolution.org/coge/). See README.txt for more information.

**Rights** (required): Information about rights held in and over the resource. Typically a Rights element will contain a rights management statement for the resource, or reference a service providing such information. Rights information often encompasses Intellectual Property Rights (IPR), Copyright, and various Property Rights. If the rights element is absent, no assumptions can be made about the status of these and other rights with respect to the resource.

> Use of the raw and processed data that is available should follow the principles of the Fort Lauderdale Agreement (https://www.genome.gov/pages/research/wellcomereport0303.pdf) and the Department of Energy's Joint Genome Institute (JGI) early release policies (http://genome.jgi.doe.gov/pages/data-usage-policy.jsf). See DATA_USE_POLICY.md for more information.

**Creator** (required): An entity primarily responsible for making the content of the resource. Examples of a Creator include a person, an organization, or a service. Typically the name of the Creator should be used to indicate the entity.

> TERRA-REF

**Contributor**: An entity responsible for making contributions to the content of the resource. Examples of a Contributor include a person, an organization or a service. Typically, the name of a Contributor should be used to indicate the entity.

**Publisher** (required): The entity responsible for making the resource available. Examples of a Publisher include a person, an organization, or a service. Typically, the name of a Publisher should be used to indicate the entity.

`CyVerse Data Commons`

**Date** (required): A date associated with an event in the life cycle of the resource. Typically, Date will be associated with the creation or availability of the resource. Recommended best practice for encoding the date value is defined in a profile of ISO 8601. For CyVerse, use the year the resource was made available.

> 2017-03-09

**Type**: The nature or genre of the content of the resource. Type includes terms describing general categories, functions, genres, or aggregation levels for content. Recommended best practice is to select a value from a controlled vocabulary (for example, the DCMIType vocabulary ). To describe the physical or digital manifestation of the resource, use the FORMAT element.

**Source**: A Reference to a resource from which the present resource is derived. The present resource may be derived from the Source resource in whole or part. Recommended best practice is to reference the resource by means of a string or number conforming to a formal identification system.

**Relation**: A reference to a related resource. Recommended best practice is to reference the resource by means of a string or number conforming to a formal identification system.

**Coverage**: The extent or scope of the content of the resource. Coverage will typically include spatial location (a place name or geographic co-ordinates), temporal period (a period label, date, or date range) or jurisdiction (such as a named administrative entity). Recommended best practice is to select a value from a controlled vocabulary (for example, the Thesaurus of Geographic Names [Getty Thesaurus of Geographic Names, http://www. getty.edu/research/tools/vocabulary/tgn/]). Where appropriate, named places or time periods should be used in preference to numeric identifiers such as sets of co-ordinates or date ranges.

**Format**: The physical or digital manifestation of the resource. Typically, Format may include the media-type or dimensions of the resource. Examples of dimensions include size and duration. Format may be used to determine the software, hardware or other equipment needed to display or operate the resource. Recommended best practice is to select a value from a controlled vocabulary (for example, the list of Internet Media Types [http://www.iana.org/ assignments/media-types/] defining computer media formats).

**Identifier**: An unambiguous reference to the resource within a given context. Recommended best practice is to identify the resource by means of a string or number conforming to a formal identification system. Examples of formal identification systems include the Uniform Resource Identifier (URI) (including the Uniform Resource Locator (URL), the Digital Object Identifier (DOI) and the International Standard Book Number (ISBN).

**Language**: A language of the intellectual content of the resource. Recommended best practice for the values of the Language element is defined by RFC 3066 [RFC 3066, http://www.ietf.org/rfc/ rfc3066.txt] which, in conjunction with ISO 639 [ISO 639, http://www.oasis- open.org/cover/iso639a.html]), defines two- and three-letter primary language tags with optional subtags. Examples include "en" or "eng" for English, "akk" for Akkadian, and "en-GB" for English used in the United Kingdom.

**Audience**: A class of entity for whom the resource is intended or useful. A class of entity may be determined by the creator or the publisher or by a third party.

**Provenance**: A statement of any changes in ownership and custody of the resource since its creation that are significant for its authenticity, integrity and interpretation. The statement may include a description of any changes successive custodians made to the resource.

**Rights Holder**: A person or organization owning or managing rights over the resource. Recommended best practice is to use the URI or name of the Rights Holder to indicate the entity.

**Instructional Method**: A process, used to engender knowledge, attitudes and skills, that the resource is designed to support. Instructional Method will typically include ways of presenting instructional materials or conducting instructional activities, patterns of learner-to-learner and learner-to-instructor interactions, and mechanisms by which group and individual levels of learning are measured. Instructional methods include all aspects of the instruction and learning processes from planning and implementation through evaluation and feedback.

**Accrual Method**: The method by which items are added to a collection. Recommended best practice is to use a value from a controlled vocabulary.

**Accrual Periodicity**: The frequency with which items are added to a collection. Recommended best practice is to use a value from a controlled vocabulary.

**Accrual Policy**: The policy governing the addition of items to a collection. Recommended best practice is to use a value from a controlled vocabulary.

nfahlgren, 2017-03-29:
The metadata can be viewed through [Data Commons](http://datacommons.cyverse.org/browse/iplant/home/shared/terraref). We can update it through the Discovery Environment when needed. If it looks good, feel free to close.

craig-willis, 2017-03-29:
@nfahlgren Sorry for the delay getting back to you on this.

Overall, what you've put together looks great.  A couple of comments:
* Subject terms: ideally Cyverse would recommend a controlled vocabulary for subject assignment.  Section 4 of http://dublincore.org/documents/dcmi-terms/ lists a few common vocabularies which are very library-centric. My initial recommendation would be to use MeSH if Cyverse doesn't have a recommendation.
* Type: This should probably be "Dataset" from http://dublincore.org/documents/2000/07/11/dcmi-type-vocabulary/ 
* Format: This is really intended for a per-file assignment, but if all of the files are in FASTQ format then you could use chemical/seq-na-fastq (I think)
* Provenance: It might be worth linking to a description of the pipeline

And on a much broader topic, what are the plans for officially publishing the genomics data -- i.e., DOI assignment? Is the intent for Cyverse to be the authoritative repository? If so, then we should probably thing about getting a DOI and assigning the appropriate contributors for data citation.

nfahlgren, 2017-03-29:
@craig-willis no worries, this is great! 

- I updated the subject terms using MeSH.
- I changed the Type to Dataset.
- I left Format out because there will be multiple types of files within the collection. I haven't done it yet, but we can apply metadata in bulk to the individual files and subfolders as well.
- I added a link to our documentation home page for Provenance in the event that the collection could have other types of data in it. We can add a link directly to the genomics pipeline for the subfolders that only contain genomics data.

The metadata records have updated on Data Commons now. In the DE I added links to the MeSH vocabulary in the Units field, which I assume is what is intended:

[image]

For publication, the raw data should be deposited in the NCBI SRA minimally. We can definitely get a DOI for the Data Commons collection as well. I opted out of that for now because we will still be adding derived data products to the collection (and possibly more raw data from related projects?). We may want to assign DOIs to sub-collections within the larger terraref collection in the event we want to keep adding more data here?

craig-willis, 2017-03-30:
@nfahlgren  The information in Data Commons looks good to me.

Thanks for the info about publishing/DOI assignment. I'm going to move this discussion to https://github.com/terraref/reference-data/issues/24 so you can close this issue.

nfahlgren, 2017-03-30:
Sounds good!

closed 2017-03-30 by nfahlgren