terraref/reference-data #106: Preliminary data package for publication

craig-willis, 2017-03-16:
This task was created based on discussions from #24. 

For the April 2017 milestone, implement preliminary data package(s) for TERRA-REF according to Nature Scientific Data requirements, for publication in Dryad or similar repository.  See [design notes](https://docs.google.com/document/d/1bwx9uWRMfft7uR_IBGkJja9dZO-Uan_5fbikz5N6Eyw/edit#heading=h.p7q8xuvex69e) for specific requirements.  In short, define the package contents, DataCite metadata, and data management policy.

Completion criteria:
* [ ] [Requirements](https://docs.google.com/document/d/1bwx9uWRMfft7uR_IBGkJja9dZO-Uan_5fbikz5N6Eyw/edit#heading=h.p7q8xuvex69e)
* [ ] Implement data package
* [ ] Implement DataCite metadata
* [ ] Define data management policy
* [ ] Stakeholder review
* [ ] Deposit with Dryad (work with curator)

craig-willis, 2017-04-20:
Had an email exchange with Dryad curator and it sounds like they are open to us depositing the data package as described.  I will put together a preliminary package for v0 and work with them to deposit.

craig-willis, 2017-06-08:
Per discussions with Rob, will use Clowder API to generate data package. Still a work in progress.

max-zilla, 2019-07-01:
@craig-willis do you still have a contact at Dryad you can share with me, or should I start anew?

craig-willis, 2019-07-01:
I've forwarded the exchange I had with the then-curator.  I know that Dryad is in the middle of a major re-launch since partnering with CDL last year. I she is still involved although not in the same role.

dlebauer, 2020-06-05:
this is done: staged in google drive for deposition at Dryad - can open new issues as necessary

closed 2020-06-05 by dlebauer