terraref/reference-data #108: Sensor metadata should be public

dlebauer, 2017-03-30:
Currently, sensor metadata are public, but require login to view collections because they are in a private space.

### Completion Criteria   

* [ ] Reorganize sensor metadata so that they are publicly available
* [ ] All links to metadata collections in documentation work

max-zilla, 2017-06-22:
I created a new public space and put pointers to these in there. The datasets were already accessible but Clowder doesn't currently support collection-level public checks so this is the most straightforward solution.

This did not change any links - @craig-willis do you know if we still need to check the documentation links, or is that something you did?

craig-willis, 2017-07-24:
Links are still working. Closing this.

closed 2017-07-24 by craig-willis