terraref/reference-data #111: API for periodic BetyDB Import

piotrfilipiuk, 2017-04-07:
### Description
New feature: I would like to be able to import entire BetyDB on a regular schedule e.g. daily.

### Details
Importing BetyDB would allow us to have the entire dataset available locally for analysis. It would also allow us to do data transformations on the entire data set.

I can see two easy solutions:
* We could simply use REST API to import the data table by table as follows: https://terraref.ncsa.illinois.edu/bety/api/beta/traits?key=$MYKEY. I am not sure however if the solution is scalable. Provided that the API supports paging, it should be doable.
* Make the PSQL dump (or equivalent) available for download.

### Completion Criteria   
* Create API to import BetyDB, or indicate that the current REST API is a way to go.

dlebauer, 2017-04-07:
@gsrohde does the api support paging?

@robkooper what do you suggest? it looks like we could set up `dump.bety.sh -u YES -l 2 -m 6` to export most of the data from terraref/bety. The only issue is that bety6 has a subset of records from bety0 (e.g. for consistent variable names). I could update the foreign ID fields in bety6 - I do not plan for the TERRA REF network to sync with the betydb.org network since there are used for very different research scopes.

robkooper, 2017-04-07:
database is already dumped and downloadable using the load.bety scripts:

https://github.com/PecanProject/pecan/blob/master/scripts/load.bety.sh#L224

So a `load.bety.sh -c -r 6` will load the latest dump to your local database.

dlebauer, 2017-04-07:
@robkooper will `-r 6` pull in all records or just those created in bety6? And we also need to add `-u YES -l 2` to the dump

gsrohde, 2017-04-07:
@dlebauer This may be moot at this point, but you can at least kind of mimic paging using the `limit` and `offset` parameters.  This may not be entirely reliable since the order in which rows are returned is not well-defined.  It does probably _work_ in spite of this.  And it wouldn't be hard to modify the API so that rows are returned in some specified order by default.

piotrfilipiuk, 2017-04-11:
Thank you for the help. Would it be possible to provide a dump in MySQL format: https://dev.mysql.com/doc/refman/5.7/en/mysqldump-sql-format.html, so that we can e.g. import the database into Google Cloud SQL?

piotrfilipiuk, 2017-04-11:
In the meantime I used the REST API to get the data table by table: https://terraref.ncsa.illinois.edu/bety/api/beta/variables?key=$MYKEY&limit=all. For now the solution is good enough, however as I mentioned I am worried about scalability once the tables get bigger.

robkooper, 2017-04-11:
Sorry we do not support mysql. The database in postgresql format, you might find software that can do the conversion.

Google SQL does support postgres (beta) see https://cloud.google.com/sql/docs/postgres/

dlebauer, 2017-04-11:
@piotrfilipiuk I thought I sent you a link to the psql dump. I've also put a copy on ROGER, so it should be avialable via Globus at sites/samples/bety6_20170405.sql.

@robkooper I edited `crontab -e -u bety` that runs dump.bety.sh so that it dumps unchecked level 2 data. Is it okay with you that I edited this file?

dlebauer, 2017-04-11:
@piotrfilipiuk using load.bety.sh as @robkooper mentioned in https://github.com/terraref/reference-data/issues/111#issuecomment-292604008 will provide the most up-to date dump

piotrfilipiuk, 2017-04-13:
Sorry, I got confused by the bety.tar.gz. I attempted the import of sites/samples/bety6_20170405.sql into Google Cloud PSQL instance, and I ran into the following issues:

1. ERROR: function utc_now() does not exist HINT: No function matches the given name and argument types. You might need to add explicit type casts.

I made a workaround by replacing the calls to utc_now() by (now() at time zone 'utc'). Would it be possible to include the function definition in the dump, or change the calls as I did? The utc_now() is not a PostgreSQL built-in.

2. ERROR: type "public.statnames" does not exist LINE 9: "statname" "public"."statnames" COLLATE "default" DEFAULT ''..

I replaced all instances with just text type

I had similar errors for
- ERROR: type "public.level_of_access" does not exist 
- ERROR: type "public.geometry" does not exist

At which point I stopped. 

Would it be possible to make the dump self contained so that it contains all dependencies that are not part of the PostgreSQL. Or maybe I am doing something wrong?

dlebauer, 2017-04-17:
@piotrfilipiuk Thank you for reporting these issues (and your patience as an alpha user). I am currently seeing if I can create a standalone psql dump. I'll let you know what I find.

Until we come up with a more stable solution, please apply for ssh access to the BETYdb (bety6) server following the first few sections of this [draft documentation](https://github.com/terraref/documentation/blob/5cc95b5fc4b2c0ef935044488dc79622a6cfe72b/accessing-betydb-with-arcmap.md). It explains how to request login access to our server and set up an ssh tunnel so that you can query the database remotely. I don't think that will be a long term solution for public access but it will work for now.

Also, regarding your comments about bety.tar.gz - did you find errors or have difficulty using the load.bety.sh script?

dlebauer, 2017-04-17:
There is a new version of the dump at sites/samples/bety6_20170417.sql that contains function and type definitions:

```sh
 grep -A 7 "Type structure for \"public\"\.\"geometry" terraref/sites/samples/bety6_20170417.sql
 grep -A 7 "Type structure for \"public\"\.\"statnames" terraref/sites/samples/bety6_20170417.sql
grep -A 6 "Type structure for \"public\"\.\"summarystats" terraref/sites/samples/bety6_20170417.sql
```

dlebauer, 2017-04-17:
@piotrfilipiuk apparently there is a bug with load.bety.sh that affects bety6 in particular https://github.com/PecanProject/pecan/issues/1355

gsrohde, 2017-04-17:
I will just go ahead and remove the problem rows from the `schema_migrations` table to see if that resolves the problem.

gsrohde, 2017-04-17:
OK, I deleted rows from the `schema_migrations` table on bety6 so that this table should match that on any other machine that is "up-to-date" schema-wise.  This _should_ get rid of any schema-mismatch errors.  But you'll have to wait until a new dump becomes available.  (I think these are done hourly.)  Please let me know if you try grabbing data from bety6 and whether or not it worked.

dlebauer, 2017-05-18:
also needed for terraref/computing-pipeline#302

rachelshekar, 2017-05-18:
@gsrohde - need to add foreign key constraints.

dlebauer, 2017-05-19:
@piotrfilipiuk 

Here are instructions for periodic import. This is using the [load.bety.sh](https://github.com/PecanProject/pecan/blob/develop/scripts/load.bety.sh) script that has been updated by @robkooper and is currently on a branch here:

https://raw.githubusercontent.com/PecanProject/pecan/357b70e21592375827e9c55bc5e1ce8e3426bd78/scripts/load.bety.sh

Now you should be able to setup an API to periodically (we often use hourly or daily) import the data using:

```
load.bety.sh -c -u -m 99 -r 0 -w https://terraref.ncsa.illinois.edu/bety/dump/bety0/bety.tar.gz
load.bety.sh -m 99 -r 6 -w https://terraref.ncsa.illinois.edu/bety/dump/bety6/bety.tar.gz
```

Note that this will overwrite any changes that you have made to rows with id fields between 1-1 billion and 6 billion+1 to  7 billion.

If this works for you please close the issue

robkooper, 2017-05-19:
**the -c flag will REMOVE the database and reinitialize** This should only be used the very first time. The second time you should not use the -c and -u flags.

rachelshekar, 2017-06-22:
@piotrfilipiuk - have you tested the periodic import?

dlebauer, 2017-08-08:
I will close this as it appears to be resolved. If further discussion is required please re-open this issue or a new one ... or find us on Slack.

closed 2017-08-08 by dlebauer