terraref/reference-data #112: How to link the BETYdb's traits information back to Globus's RGB data?

luckycatcs, 2017-04-11:
### Description

Hi team,

We pulled some traits/variables/treatments data from BETYdb and we would love to link them back to the RGB data available on Globus. The goal is to train a model "f(Globus RGB data) -> traits" to predict traits. We checked the schema and relation of the BETYdb TERRA-REF database but didn't find any way to do that. Could you please give us some advice? Thanks!

### Completion Criteria   
Some examples or documentations which demonstrate how to link the data.

dlebauer, 2017-04-12:
### Part 1: Query date+time and location from BETYdb 

Clowder and BETYdb can be cross-queried using the date of observation and the location.

From BETYdb, you can get the time of a record from the traits table. Let's start with the id of the record in the traits table:

In SQL:

```sql
select date, site_id, ST_AsGeoJSON(geometry), mean from traits join sites on traits.site_id = sites.id where traits.id = 6000091915;
```

* date: 2016-10-06 05:00:00	
* geometry: {"type":"Polygon","coordinates":[[[-111.975032369538,33.0763972541809,353],[-111.975015978712,33.0763972619415,353],[-111.975015956477,33.0763640256532,353],[-111.975032347404,33.0763640178919,353],[-111.975032369538,33.0763972541809,353]]]}

Via API 

```sh
https://terraref.ncsa.illinois.edu/bety/api/beta/traits?id=6000091915&key=
## then use the returned site_id to query the sites table
https://terraref.ncsa.illinois.edu/bety/api/beta/sites?id=6000003804&key=
```

@jterstriep and / or @Zodiase could you please provide Part 2, how to use the date from https://terraref.ncsa.illinois.edu/bety/api/beta/traits?id=6000091915&key= and the polygon from https://terraref.ncsa.illinois.edu/bety/api/beta/sites?id=6000003804&key= to find all images from the StereoTop camera with an intersecting field of view collected on the same day?

robkooper, 2017-04-13:
You can use the following query:

```
 curl -u username:password \
   'https://terraref.ncsa.illinois.edu/clowder/api/geostreams/datapoints?geocode=-111.975032369538%2C33.0763972541809%2C-111.975015978712%2C33.0763972619415%2C%20-111.975015956477%2C33.0763640256532%2C%20-111.975032347404%2C33.0763640178919%2C%20-111.975032369538%2C33.0763972541809%2C&format=json'
```

dlebauer, 2017-04-13:
@robkooper that will work once we populate geostreams with the plot level subset images. Now we should have field of view at time of capture for each of the unprocessed geotiffs.
 
@max-zilla or @craig-willis is there a way to query all of the field of view polygons from the Clowder MongoDB along with their associated files (paths on ROGER)? Then, e.g., @luckycatcs could import these into a geospatial database and perform the queries there (and ideally post the code used to do this ... here). This would be particularly useful in the short term until we get the plot-level clipped images, but may be useful even in the long term for people who want the raw rather than stitched/clipped images

this is a use case for https://github.com/terraref/computing-pipeline/issues/231

dlebauer, 2017-05-03:
@luckycatcs We have produced our first full-field stitched data product from the RGB images.

* ROGER: /gpfs/largeblockFS/projects/arpae/terraref/sites/ua-mac/Level_1/fullfield/2017-04-27
* Globus #TERRAREF endpoint: /ua-mac/Level_1/fullfield/2017-04-27/
* terraref.ndslabs.org applications: /data/terraref/sites/ua-mac/Level_1/fullfield/2017-04-27

You will find

```
 100G May  3 16:59 stereoTop_fullfield.tif     ## geotiff file
 13M  May  3 08:39 stereoTop_fullfield.VRT  ## VRT file
 1.1M May  3 08:30 stereoTop_tifList.txt       ## list of all source files used to create VRT+geotiff
```

The VRT is a virtual geotiff. The geotiff was produced with `gdal_translate stereoTop_fullfield.VRT stereoTop_fullfield.tif` and is easiest to work with. In practice we may only generate the geotiff on demand. See related issues https://github.com/terraref/computing-pipeline/issues/126 (generating this stitched mosaic) and https://github.com/terraref/computing-pipeline/issues/265 (generating plot level subsets).

Given this geotiff file, it should be possible to subset the image given the plot boundaries from BETYdb. The current Season 4 plots should be the same as (within a few cm) the Season 1 plots, so you can use either https://terraref.ncsa.illinois.edu/bety/api/beta/sites?key=9999999999999999999999999999999999999999&limit=none&name=~Season+1 or `sql select name, geometry from sites where name like '%Season 1%';` if you either have a copy of the betydb database or access to the production database via ssh tunnel  as described in our [documentation for connecting to the database w/ GIS software](https://terraref.gitbooks.io/terraref-documentation/content/accessing-betydb-with-arcmap.html). 

We are also in the process of adding the geospatial meta-data that will allow queries of the clowder (image) database like the one Rob posted above (https://github.com/terraref/reference-data/issues/112#issuecomment-293954102).

dlebauer, 2017-09-28:
* can query geostreams given a plot bounding box
* can use / extend @jterstriep s API

Assigning to Jeff: please write a new feature ticket for what can be implemented and documented for Nov data release

dlebauer, 2017-09-28:
Need to have some code samples that can be put in the tutorials repository.

dlebauer, 2020-06-05:
a sqlite database to facilitate this will be implemented here: https://github.com/AgPipeline/issues-and-projects/issues/146

closed 2020-06-05 by dlebauer