terraref/reference-data #114: Plots and rows

rachelshekar, 2017-04-13:
from Nadia:

Plots can be defined as the field area that is occupied by a single genotype in a given rep.  We can do 1 row, 2 row, 3 row, 4 row plots (whatever we want) depending on how many rows of one genotype we want in a particular area. This will change from experiment to experiment. 

It is best to keep the data separated by row, but we need to have the ability to combine row information in order to generate plot-level data. 

Which sensors will need data to be clipped by row?

This affects 
https://github.com/terraref/computing-pipeline/issues/210
https://github.com/terraref/reference-data/issues/86
https://github.com/terraref/computing-pipeline/issues/265
https://github.com/terraref/computing-pipeline/issues/183
https://github.com/terraref/computing-pipeline/issues/186
https://github.com/terraref/reference-data/issues/60

rmgarnett, 2017-04-13:
#60: I think the right thing to do here is to do all processing/storage at the "1-row" plot level, then provide the ability to aggregate into logical divisions afterwards.

NewcombMaria, 2017-04-13:
A note on terminology that may help avoid confusion:  the one-row unit is a 'subplot' when the area occupied by a single genotype (plot) is 2 or more rows.  The subplots (1-row) are then aggregated into logical units which could be the plot (single genotype experimental unit).

rickw-ward, 2017-04-13:
I suggest we adopt the logic and terminology used by the Breeding Management System described here:  https://www.integratedbreeding.net/283 .   This is a project level question that I request be answered asap.

This is a departure for me because I have used "Pass" in place of "Column", but old dogs can learn so I will move to the BMS approach.    This is advantageous since TERRA-REF (MAC, Kansas State) aim to employ BMS- which is supported by BMGF and other big donors.

________________________________________________
**Row, Range, & Plot Details
Field Map Definitions:**

Row: Planting row; if only one row per plot is specified, row is equivalent to column
Column: Vertical position of plots, width determined by the number of rows per plot
Range:  Horizontal position of plots
Plot: Specified by column (vertical position) and range (horizontal position), may contain 1-8 planting rows 

[image]

## **Below: 2D representation of rows, columns, and ranges; each row has a unique range/row, and each plot has a unique range/column.  For the gantry, this coordinate system would be 54 ranges x 32 rows.  For two row plots, there are 16 plots per range.**   

[image]

## **Below: table/ledger (Field Book) view (does not drill down to "rows" within plots). The name of a plot is a binomial ID composed of Column/Range, and each also has a sequential ID ("Plot No." in instance below.)  This table below does not show the rows within a plot as separate objects/lines, but I believe we could do that**.

[image]

rachelshekar, 2017-04-20:
Default search would be plot.  Row (subplot) would also be available.
Row length may change from season to season.

craig-willis, 2018-06-13:
This is a stale issue with no comments in last 2 months. Please create new issues if work remains or put this content in the documentation.

closed 2018-06-13 by craig-willis