terraref/reference-data #116: Upload derived data products for BAP resequencing

nfahlgren, 2017-04-25:
### Description

Raw data for whole-genome resequencing of 384 accessions from the BAP has been uploaded to both the TERRA-REF and CyVerse data stores. Next we need to upload derived data products when they are ready. Minimally we will need the combined hapmap file so that the same reference hapmap can be used by all teams. This data would be uploaded to `/hudson-alpha/derived_data/` on the TERRA-REF platform, and `/iplant/home/shared/terraref/sites/hudson-alpha/derived_data` on CyVerse.

### Completion Criteria   

- [ ] Upload hapmap file
- [ ] Upload individual cleaned VCF files?
- [ ] Other useful intermediate files that should be shared?

NewcombMaria, 2017-04-25:
Phenotyping on the BAP will be starting in Maricopa soon (both sensor and field measurements).  This is a great time to connect the unique BAP identifiers used for resequencing data with the phenotypic data that will be available soon (issue #65 ).  Is there a way to get a list of the BAP identifier names used for resequencing genomic data uploaded to iplant and elsewhere?  I'll use the same identifiers for phenotypic field measurement data files.

nfahlgren, 2017-04-26:
@NewcombMaria great question! I talked to @phloz about what identifiers the resulting hapmap will use and he added them to the sample spreadsheet we have: https://docs.google.com/spreadsheets/d/1ntnps0u0c8O4xxrv_VVqfxDkpmppCGdFIIWLMmRCeSM/edit?usp=sharing

In that table the `identifer` column is how Phil will label the genotype data. Note that some of the samples are duplicates and will be merged in the final hapmap (e.g. PI564163 refers to both PI 564163 and BTx623).

Does that work for you?

NewcombMaria, 2017-04-26:
Thank you @nfahlgren.  I looked at the identifier column in the sample spreadsheet and this will make it easy to match genotype/phenotype data.  Thanks!

nfahlgren, 2017-10-26:
gVCF and hapmap files are now uploaded to CyVerse at:

`/iplant/home/shared/terraref/genomics/derived_data/bap/resequencing/danforth_center/gvcf`

and

`/iplant/home/shared/terraref/genomics/derived_data/bap/resequencing/danforth_center/hapmap`

closed 2020-06-05 by dlebauer