terraref/reference-data #117: Upload raw and derived data products for GBS on RILS

nfahlgren, 2017-04-25:
### Description

As with issue #116, I am creating this issue as a reminder to upload raw and derived data products for genotyping of the sorghum RIL populations. Like the resequencing data, I think we can upload the data to TERRA-REF using Globus, and minimally we will need the combined hapmap file so that the same reference hapmap can be used by all teams. Because KSU will be uploading both genomic data and image data, we will need a somewhat different structure than we used for the BAP resequencing. I suggest we upload the raw data to `/ksu/raw_data/gbs` and the derived data to `/ksu/derived_data/gbs`. Since there are two populations, subfolders might be needed?

### Completion Criteria   

- [ ] Upload raw data (FASTQ, key files, etc.)
- [ ] Upload hapmap file
- [ ] Other useful intermediate files that should be shared?

dlebauer, 2017-04-26:
Does this go on Roger, Cyverse, or both?

nfahlgren, 2017-04-26:
@dlebauer I think just Roger for now. If it's okay to put up on the CyVerse Data Commons as well, let me know and I can transfer a copy from Roger.

dlebauer, 2017-04-26:
* uploaded 2 FASTQ files and one key file to ksu/genomics/raw_data/ril_gbs/
* uploaded vcf hapmap file to ksu/genomics/derived_data/ril_gbs/ 

@zhenbinHU does not think that there are other useful intermediate files, but said there will be more derived data.

But out of curiosity, does it make sense to have these divided into sites/ksu vs sites/hudson-alpha, or could we combine everything in a single genomics/ folder?

nfahlgren, 2017-04-26:
@dlebauer combining them is fine with me, there's not really a need to keep the overall genomic data separate. So we could reorganize the data on CyVerse also to be /terraref/genomics/...

nfahlgren, 2017-04-26:
@dlebauer want me to move things around and update the documentation?

dlebauer, 2017-04-26:
That would be great - if you want to organize on CyVerse I can reorganize on Roger, or if you tell me how to do it on roger I can do it there first.

max-zilla, 2017-04-26:
Started uploading to Clowder as well.
https://terraref.ncsa.illinois.edu/clowder/datasets/5900f3f34f0c38c084c862dd

nfahlgren, 2017-04-26:
From the root folder on Globus (sites I think), we could do:

```
/genomics
    /raw_data
        /ril_gbs
        /bap_resequencing
    /derived_data
        /ril_gbs
        /bap_resequencing
```

Within the bap_requencing we could put all the data together (no year1/year2 subfolders)

Thoughts?

dlebauer, 2017-04-26:
That sounds good to me ... is ril:gbs equivalent to bap:resequencing? Or might we have a ril_resequencing and / or bap_gbs one day (in which case maybe ril/gbs and bap/resequencing)?

nfahlgren, 2017-04-26:
That sounds good. RIL/BAP is the population type and GBS/resequencing is a sequencing strategy, so we could theoretically apply the strategies to either population.

dlebauer, 2017-04-27:
@nfahlgren OK I've rearranged as suggested above. 

```sh 
# https://askubuntu.com/a/431252/2487
$ find genomics -type d | sed -e "s/[^-][^\/]*\//  |/g" -e "s/|\([^ ]\)/|-\1/"
.
  |-raw_data
  |  |-bap_resequencing
  |  |-ril_gbs
  |-derived_data
  |  |-bap_resequencing
  |  |-ril_gbs
```

I put both the data use policy and README under genomics/. 

Could you please modify the README.txt and documentation as necessary?

@max-zilla if the uploads to globus have pointers, I think they will need to be re-uploaded or fixed. Sorry about that.

nfahlgren, 2017-04-27:
@dlebauer wanted to double check, did we want to lay it out like the following?

```
  |-raw_data
  |  |-bap
  |  |  |-resequencing
  |  |-ril
  |  |  |-gbs
  |-derived_data
  |  |-bap
  |  |  |-resequencing
  |  |-ril
  |  |  |-gbs
```

dlebauer, 2017-04-27:
Fixed:

```sh
[genomics]$ find ../genomics -type d | sed -e "s/[^-][^\/]*\//  |/g" -e "s/|\([^ ]\)/|-\1/"
  |-genomics
  |  |-raw_data
  |  |  |-ril
  |  |  |  |-gbs
  |  |  |-bap
  |  |  |  |-resequencing
  |  |-derived_data
  |  |  |-ril
  |  |  |  |-gbs
  |  |  |-bap
  |  |  |  |-resequencing
```

nfahlgren, 2017-04-27:
@dlebauer awesome, thanks!

max-zilla, 2017-04-27:
@dlebauer deleted the existing file pointers and re-uploaded for GBS RIL this morning.

nfahlgren, 2017-04-27:
@dlebauer can you remove these two files, I will replace with one merged file:

`/genomics/raw_data/bap/resequencing/terraref_year1_sorghum_bap_samples.txt`
`/genomics/raw_data/bap/resequencing/terraref_year2_sorghum_bap_samples.txt`

nfahlgren, 2017-04-27:
@dlebauer I also started an upload to update the README but don't have permission to replace the file.

dlebauer, 2017-04-27:
I've given you write permission to the /genomics directory.

nfahlgren, 2017-04-27:
@dlebauer reorganization of the data on CyVerse is done. I submitted pull request [#150](https://github.com/terraref/documentation/pull/150) to update the documentation for the BAP data. I did not add information for the RIL GBS data.

dlebauer, 2017-04-28:
@nfahlgren I see the bap but not the ril data on Cyverse ... http://datacommons.cyverse.org/browse/iplant/home/shared/terraref/genomics

dlebauer, 2017-04-28:
@gpmorrisksu can we share the gbs data with alpha users via Cyverse? That is our public portal for genomics data (http://datacommons.cyverse.org/browse/iplant/home/shared/terraref/genomics). 

It is already available to alpha users via Globus and Clowder. 

Both places present the data use policy, but the alpha users have had to affirm their willingness to comply.

dlebauer, 2017-05-03:
@nfahlgren could you copy this to CyVerse but keep the folder private as I've done with the bap/resequencing folders?

nfahlgren, 2017-05-04:
@dlebauer I downloaded the files locally and am gzipping them now, then will upload them to CyVerse.

nfahlgren, 2017-05-05:
@dlebauer the data are now uploaded to CyVerse here: `/iplant/home/shared/terraref/genomics/raw_data/ril/gbs`

@max-zilla the Clowder component of this issue is complete right? If so I think we can close this issue now, or is there anything else we need to do as part of this issue @dlebauer?

max-zilla, 2017-05-05:
@nfahlgren yep. as far as I'm concerned, close away.

closed 2017-05-05 by nfahlgren