terraref/reference-data #118: Merge plant_height and panicle_height

dlebauer, 2017-04-28:
In the crop ontology, plant height is defined as the "Length of the main stalk from the base of the stalk to the tip of the panicle." http://www.cropontology.org/terms/CO_324:0000088/

We currently have 1040 records of 'panicle height' and 915 of 'plant height'. I would like to merge these variables, but want to avoid loosing information such as the fact that a panicle was present at the time of measurement. 

@NewcombMaria does it make sense to merge these two variables?

### Completion Criteria   

* [ ] decide wheter to merge plant_height and panicle_height
* [ ] to merge these variables:

```sql
update traits set variable_id = 6000000005 where variable_id = 6000000017;
delete from variables where id = 6000000017;
```

gpmorrisksu, 2017-04-28:
> Currently we have separate variables for plant height and panicle height
>
>   *   plant_height<terraref.ncsa.illinois.edu/bety/variables/6000000005>
>   *   panicle_height<terraref.ncsa.illinois.edu/bety/variables/6000000017>
>   *   spike_height<https://terraref.ncsa.illinois.edu/bety/variables/6000000008>
>
> panicle is sorghum specific
> spike is a generic term for height to top of reproductive organ.

Are you referring to KSU data or Maricopa data or both?

------------------------------------
Geoff Morris, Assist. Professor
Kansas State Univ., Agronomy
3004 Throckmorton<x-apple-data-detectors://6/0> Plant Sci.
Manhattan KS, 66506<x-apple-data-detectors://6/0>
E-mail: gpmorris@k-state.edu<mailto:gpmorris@k-state.edu>
Web:http://www.morrislab.org<http://www.morrislab.org/>
Office: 785-532-3397<tel:785-532-3397>
Cell: 312-909-1330<tel:312-909-1330>
Skype/Google: morris.geoff.p

In sorghum panicle_height = plant_height.
For some crops (like corn) spike_height < plant_height

@NewcombMaria<https://github.com/NewcombMaria> and @gpmorrisksu<https://github.com/gpmorrisksu> does it make sense to merge plant_height, panicle_height, and spike_height?

Completion Criteria

  *    decide whether to merge plant_height, panicle_height, spike_height
  *    decied whether to delete panicle_height and spike_height
  *    merge these variables:

update traits set variable_id = 6000000005 where variable_id = 6000000008;
delete from variables where id in(6000000008, 6000000017);

dlebauer, 2017-04-28:
Both ... Ideally we will have the same names for the same property being
measured.

gpmorrisksu, 2017-04-28:
> E-mail: gpmorris@k-state.edu<mailto:gpmorris@k-state.edu><mailto:gpmorris@k-state.edu>

The crop ontology definition is a bit off base because it assumes (for no good reason) that phenotyping is post-heading.

Most of our measurements would be from material that hadn't flowered or would never flower.

Our measurements should be at the top of the canopy, as judged by the data collector.

------------------------------------------------------------------------------------------------------------------------
Geoff Morris, Assistant Professor
Department of Agronomy | Kansas State University
3004 Throckmorton Plant Science Center | Manhattan KS, 66506
E-mail: gpmorris@k-state.edu<mailto:gpmorris@k-state.edu> | Web: http://www.morrislab.org<http://www.morrislab.org/>
Office: 785-532-3397 | Cell: 312-909-1330 | Skype/Google ID: morris.geoff.p

NewcombMaria, 2017-04-29:
Thanks @dlebauer for questions on this topic.  It will be useful to settle on common variable names.  

Going forward with measurements on the BAP, as @gpmorrisksu stated, most height measurements from now on at Maricopa will be on material that has hasn't yet flowered and may never flower, and it makes sense to measure 'canopy height' (to top of canopy excluding panicle if one exists).  The 2 Maricopa sorghum seasons in 2016 were different with more grain types in the field.  I think it's important to retain the information from those 2 seasons that helps the image analysts know if the height includes a panicle or not for each plot.  Potentially there could be image segmentation into leaves (canopy) and panicles, and heights determined for each.  If it seems best to merge plant height = panicle height, then to retain all information we need to restrict the definition of 'plant height' only to post-heading measurements for 'base of stalk to tip of panicle', otherwise the measurement is 'canopy height'.  With grain types, at any single date during heading time, plots that have not headed will have data for canopy height alone and plots that already have headed will have data for both canopy height and plant height.  If that seems better than canopy height and panicle height, it will work to retain all info if plant height is restricted to only post-heading.  

It seems more descriptive to use the variables canopy height and panicle height.  However, I understand the need to standardize variables and think about long-term use of the database.

craig-willis, 2018-06-13:
Stale issue. No comments in last 2 months. If necessary, create new issues based on this discussion.

closed 2018-06-13 by craig-willis