terraref/reference-data #121: Upload field and plot boundaries from 2016 KSU

dlebauer, 2017-04-28:


dlebauer, 2017-06-01:
Field plot boundaries:

[KSU_2016_TERRA_fieldmap.csv.txt](https://github.com/terraref/reference-data/files/1045261/KSU_2016_TERRA_fieldmap.csv.txt)

dlebauer, 2017-06-07:
@zhenbinHU could you provide these plot boundaries as a dump directly to a csv, this will make it easier to import since we are both using PostGIS? 

something like

```sql
COPY (select sitename, geometry from sites) TO ~/ksu_sites.csv;
```

should do the trick. Otherwise someone needs to a) transform these to WGS84 projection.

zhenbinHU, 2017-06-07:
@dlebauer The field plot boundaries was uploaded in csv format, I don't what happened to it. It will become to csv format if you can remove the .txt in file name (KSU_2016_TERRA_fieldmap.csv.txt).

dlebauer, 2017-06-07:
@zhenbinHU I renamed it .txt. But the problem is _not_ the file format, it is that it will be non-trivial to import data from this format. Certainly doable, but it would be much easier if the fields were `sitename, geometry` as the data are stored in PostGIS. Since (as I understand it) these data come from your PostGIS database, I suspect it will be much easier to just copy the data from the database to csv.

dlebauer, 2017-06-15:
From Kevin:
> The CRS of this field map shows as "EPSG:32614, WGS 84 / UTM zone 14N" in GIS. 

plot definitions provided by Kevin in WKT w/ sitename and cultivar name are here: https://docs.google.com/spreadsheets/d/1APkedH70txKfvUknvgiqDTsdLWC2uriTj5dXIrHNY74/edit#gid=1725314081

Issues with these data:

1. Sitenames provided by Kevin are of the format `TR16_Rg1_P1`; the corresponding plot name provided by Zhenbinn and inserted into the database was `TR_R01_P1`. The differences are 
   * `R` --> `Rg`
   * Range 1 is called `Rg1` vs `Rg01`
2. The RIL genotype should not have the underscore so I renamed all `PI_xyz` to `PIxyz` 
2. All of the 843 plots existed in the database
2. Of the 843 plot-genotype combinations provided, only 529 had matching genotypes in the database. 
   *  There are 174 plots with no associated cultivar. These are listed here: https://docs.google.com/spreadsheets/d/1APkedH70txKfvUknvgiqDTsdLWC2uriTj5dXIrHNY74/edit#gid=1725314081
3. Plots have no elevation values. I suspect these would be important when handling 3D data.

dlebauer, 2017-06-15:
Also, to coerce these polygons into PostGIS I had to wrap them thus:

* `ST_Force3d(ST_Transform(ST_GeomFromText('Polygon( x y x y x y x y x y)', 32614),4326))

closed 2017-06-22 by dlebauer