terraref/reference-data #124: Add KSU stitched geotiff images to Clowder

dlebauer, 2017-04-29:
Files from sites/ksu/Level_1/dsm_fullfield
sites/ksu/Level_1/rededge_fullfield should be uploaded to https://terraref.ncsa.illinois.edu/clowder/datasets/5900e46f4f0ce950158ca4f2?space=5900d9504f0ce950158c5792

## Completion Criteria

* [ ] All files from above folders available via Clowder, with file metadata extracted

zhenbinHU, 2017-05-08:
I am trying to transfer raw image data to Globus, but seems I don't have permission?
.//2016-08-22T17-49-35Z_BGREN_16ASH-TERRA.tif: Read-only file system.

dlebauer, 2017-05-08:
@zhenbinHU I have given you access to write to Globus. However, for the long term it will likely be easier to set up a Globus file mover like we have at UA. This makes it so that files are automatically transferred and registered with Clowder when they are put in a directory on your end. 

@jdmaloney can you help @zhenbinHU set this up?

jdmaloney, 2017-05-08:
I'll echo @dlebauer 's sentiments, with write access to the endpoint you have the power to delete (please don't) so tread carefully, few accounts have this access.  

The transfer code from UA will have to be modified a bit as KSU is using a database to track files (instead of the xfer log system at MAC).  Some progress has been made on that but it is not finished yet, I'll differ to @max-zilla on that per his meeting with KSU last week.  The good part is that since there really isn't much data from KSU on ROGER (and not sure if the few files that are in those directories are in clowder).  I can just scan that folder before we cut over to the automated pipeline to get a list of files to inject into clowder.

jdmaloney, 2017-05-18:
Updated the check boxes on this to reflect what has been completed.  Level_2 looks to be not completed yet (empty directory on ROGER), but raw data is transferred.  @max-zilla You'll probably want to wait for Level_2 to be done and register it all with Clowder at once, but up to you.  

As far as I know my part of this issue is complete.  Let me know if I'm mistaken.