terraref/reference-data #125: Yield data components and analyses - Maricopa

NewcombMaria, 2017-05-04:
Prior to uploading the yield data from Maricopa sorghum seasons 1 and 2 (July 2016 harvest and November 2016 harvest), some questions have been raised:

**Which variables are of interest to BETYdb users?**  

- data components for aboveground fresh biomass, aboveground dry biomass, aboveground moisture content, or simply final values?  
- associated data as potential covariates (measured gaps, bird damage, etc)?

**How to handle calculations, SAS code?**  

**Version control:**  there have been questions about how much time to spend before uploading to BETYdb to ensure the best quality data, versus meeting timely deadlines.  Is there a method for correcting errors after uploading?

Other questions from @JeffWhiteAZ ?

rachelshekar, 2017-06-01:
@dlebauer

dlebauer, 2017-06-15:
> Which variables are of interest to BETYdb users?

* aboveground dry biomass
* aboveground moisture

* associated data as potential covariates (measured gaps, bird damage, etc)?

we can add these as covariates if relevant. 

* How to handle calculations, SAS code?

Best woulld be to write up the protocol on protocols.io and put the SAS code in the terraref/reference-data repository. Then use the citation / doi from protocols.io in the methods field of the database.

> Version control: there have been questions about how much time to spend before uploading to BETYdb to ensure the best quality data, versus meeting timely deadlines. Is there a method for correcting errors after uploading?

Absolutely! part of having alpha, beta, and v1 data is to give time to review and revise data along the way.

NewcombMaria, 2017-06-15:
@dlebauer we had a meeting in mid May and made some decisions about how the yield data needed to be recalculated.  I now have a revised file and was planning to upload soon.  The data from the spreadsheet https://docs.google.com/spreadsheets/d/14mO_1-en9dj8q_dQeXJOKEB0sYKlF_LR__rEIqg5-kw/edit#gid=151802899 are not current.

dlebauer, 2017-06-22:
@NewcombMaria can this issue be closed?

NewcombMaria, 2017-07-10:
Yes this can be closed.  All of the 2016 season1 and season2 harvest yield data and covariates are now uploaded.  Still need to write protocols and methods descriptions complete enough for protocols.io, but protocols are part of a larger topic/issue.

closed 2017-07-12 by dlebauer