terraref/reference-data #126: Upload MAC Season 4 (First Sorghum crop of 2017) metadata

dlebauer, 2017-05-05:
### Completion Criteria

* [ ] plot definitions
* [ ] experiments_sites relationships
* [ ] cultivars_sites relationships 
* [ ] planting date

dlebauer, 2017-05-05:
Inserted plot definitions for seasons 3 and 4
* [season_4_sites.txt](https://github.com/terraref/reference-data/files/980226/season_4_sites.txt)
* [season_3_sites_.txt](https://github.com/terraref/reference-data/files/980234/season_3_sites_.txt)

NewcombMaria, 2017-05-05:
@dlebauer for seasons 3 and 4 there are 14 plots (sites) for each Range, instead of 7 (season 2 was the only season with the 4-row plots or 7 site names per Range).  For both season 3 (durum wheat) and season 4 (sorghum summer 2017), each Range has 14 plots (2-row plots).  Apparently the BRAPI system uses 'Column' instead of 'Pass'.    The season 3 and season 4 site names for one Range should be as follows:  
MAC Field Scanner Season 4 Range 3 Column 1  
MAC Field Scanner Season 4 Range 3 Column 2
MAC Field Scanner Season 4 Range 3 Column 3
MAC Field Scanner Season 4 Range 3 Column 4
MAC Field Scanner Season 4 Range 3 Column 5
MAC Field Scanner Season 4 Range 3 Column 6
MAC Field Scanner Season 4 Range 3 Column 7
MAC Field Scanner Season 4 Range 3 Column 8
MAC Field Scanner Season 4 Range 3 Column 9
MAC Field Scanner Season 4 Range 3 Column 10
MAC Field Scanner Season 4 Range 3 Column 11
MAC Field Scanner Season 4 Range 3 Column 12
MAC Field Scanner Season 4 Range 3 Column 13
MAC Field Scanner Season 4 Range 3 Column 14

dlebauer, 2017-05-05:
@NewcombMaria This is the current layout for the first 3 ranges 

[image]

Does that look correct? I assume that the 14 plots per range are excluding the two border columns. I've followed the map in the [Season 4 google spreadsheet](https://docs.google.com/spreadsheets/d/1iUTkyU5P3IfohQg-sLq_ETWvaWBHF5Nvchx3iI8C8Qw/edit#gid=1165578389). And I've kept all 16 columns and 54 ranges (including border rows) since the border rows still show up under the gantry.

NewcombMaria, 2017-05-06:
Thanks @dlebauer !  Keeping all 54 ranges and 16 columns is probably a good idea.  In your labeled image of the layout I don't see Ranges 53 and 54.  There's been talk about placing targets in Range 53 so would be useful to include site names for Range 53.  It's correct that the 14 plots per Range are the experimental plots and exclude the border rows (Columns 1 and 16 are borders on West and East sides).   Ranges 1 and 2 are borders on the South end, and Ranges 53 and 54 are borders on the North end.  The sensor box is not able to scan Ranges 1 and 54, but it can reach most of Ranges 2 and 53.  There are 700 experimental plots (2-row plots) withing Ranges 3-53, and Columns 2-15 (50 * 14).  The border Columns and Ranges are sometimes used for tests or destructive samples so it's useful to give them site names.  Thanks!

dlebauer, 2017-05-08:
@NewcombMaria sorry - we do have all of the ranges through 54. These were just off the screen when I created the images above.

dlebauer, 2017-05-24:
Need to update plot definitions https://docs.google.com/spreadsheets/d/1eQSeVMPfrWS9Li4XlJf3qs2F8txmddbwZhjOfMGAvt8/edit#gid=1765508846

dlebauer, 2017-05-25:
I've entered 
* cultivars https://docs.google.com/spreadsheets/d/1iUTkyU5P3IfohQg-sLq_ETWvaWBHF5Nvchx3iI8C8Qw/edit#gid=1160834075
* experiment_sites https://docs.google.com/spreadsheets/d/1iUTkyU5P3IfohQg-sLq_ETWvaWBHF5Nvchx3iI8C8Qw/edit#gid=552329304
* sites_cultivars https://docs.google.com/spreadsheets/d/1iUTkyU5P3IfohQg-sLq_ETWvaWBHF5Nvchx3iI8C8Qw/edit#gid=404544108 entered

dlebauer, 2017-05-25:
@NewcombMaria the cultivars list you gave me is here: https://docs.google.com/spreadsheets/d/1iUTkyU5P3IfohQg-sLq_ETWvaWBHF5Nvchx3iI8C8Qw/edit#gid=1160834075

Can you confirm that these are the correct names? It seems that some of these varieties were grown in Season 1 but we gave them new names. If we should use PI numbers or such, please provide a table of oldname, newname so that I can update the database

NewcombMaria, 2017-05-26:
@dlebauer thanks for your question about cultivars.  It's best to use the PI numbers.  I added a column for -'name_new' in the spreadsheet:  https://docs.google.com/spreadsheets/d/1iUTkyU5P3IfohQg-sLq_ETWvaWBHF5Nvchx3iI8C8Qw/edit#gid=1160834075

The only changes were LEOTI (old name) to PI_641825 (new name) and ATLAS (old name) to PI_641807 (new name).  I changed both entries to BAP in the ecotype column.  We also had 'ATLAS' in season1 but that ATLAS was PI_586537 (season4 ATLAS is PI_641807).  I found this confusing so I checked in with the Clemson group who provided the seed and received this response:  "Hello again Maria:  PI_586537 and PI_641807 both represent Atlas. They are just different seed lots that represent Atlas."  

Does the cultivar list for Season1 BAP accessions have PI numbers or common names?  PI numbers would be best.

NewcombMaria, 2017-05-26:
@dlebauer there are different ways to write PI numbers.  According to @nfahlgren the genomic data file uses identifiers written in the format PI564163 (instead of PI_564163).  Do you want to establish a standard format now for BETYdb?

dlebauer, 2017-05-26:
Yep let's use the standard format for BETYdb it should be easy to fix.

NewcombMaria, 2017-05-29:
@dlebauer I added a new column in the spreadsheet for the cultivar name in the same format as the genomic identifier, for example PI329465.

NewcombMaria, 2017-07-08:
@dlebauer with thanks and credit due to Andy French we have progress on shape files (2017 season4) for each East/West row subplot of all of the 864 plots (including border plots) using the scanner coordinate system.  The polygons are defined using the file with X/Y coordinates that Stuart generated from alignment with the sensor box and field stakes that Pedro and I measured out.  Andy also converted the polygons to UTM coordinates for use with the UAV data, and we are in process of quality-checking the UTM coordinate files.  Andy asked me to get feedback on how to handle the projection component for the scanner coordinate files.  Andy generated 4 files, one of which is .prj file, and this part of the shape file is currently set with it's own origin but Andy wants input on how to handle (or even if to include) a projection component.  Do you know who could answer this?  Also @max-zilla created a full-field flirIR image (closed computing-pipeline issue #327) and it would be great if the full-field image could be used to quality check these shape files.

NewcombMaria, 2017-07-08:
@dlebauer I put the 4 shape files (scanner coord) that Andy French generated in the Google folder 'Sorghum season4 2017 shape files plot polygons:
https://drive.google.com/drive/folders/0B1OR5-zeXagneWVJNVk1S05kQlk

dlebauer, 2017-07-11:
I added the UTM shapefiles to the same directory on google drive (https://drive.google.com/drive/folders/0B1OR5-zeXagneWVJNVk1S05kQlk). 

@jterstriep can you help me upload these to BETYdb? We will want to replace the existing plot boundaries, but in the shape files, each current two-row plot is represented by two polygons (one to the east and one to the west) . 

[image]

Here is an illustration. Red polygons are the current definitions and purple are the new definitions. Two purple boxes correspond to each red box, what we need to do is update the geometries in BETYdb thus:

```sql
update sites set geometry = st_union(select geometry from shapefile where Plot_name = 'MAC Field Scanner Season 4 Range 11 Column 2') where sitename = 'MAC Field Scanner Season 4 Range 11 Column 2';
```

We should go ahead and keep the subplots as well, thus:
```sql
insert into sites sitename = (select Subplot_na from shapefile), geometry = (select geometry from shapefile) ;
```

dlebauer, 2017-07-11:
I've inserted the combined plots for Season 4 

1. I imported the shapefile to PostGIS using the "PostGIS shapfile uploader"
2. transformed geometry using [`st_force3D(st_transform(st_union`](https://gist.github.com/dlebauer/919dce0b269bfbed72dc4a567f1d97cc#file-select_transform_geometry-sql)
3. generated update statements in [google docs](https://docs.google.com/spreadsheets/d/1iUTkyU5P3IfohQg-sLq_ETWvaWBHF5Nvchx3iI8C8Qw/edit#gid=965395161) and [as sql](https://gist.github.com/dlebauer/919dce0b269bfbed72dc4a567f1d97cc#file-update_sites_maricopa_season_4-sql)
4. Generated insert statements for subplots https://gist.github.com/dlebauer/040b8d7f54872ae688a86e9bdbd87976

NewcombMaria, 2017-07-11:
@dlebauer if it's helpful, Andy figured out how to export the attribute table with plot centroids in both Lat/Long and UTM, with the site/cultivar/subplot information.  I put an excel file in the google docs directory:  https://docs.google.com/spreadsheets/d/1AzHwuTkMoMJCJYy2hUNEJuM3YRJR7U1zORp4dMraNMw/edit#gid=1613139443

dlebauer, 2017-07-11:
@NewcombMaria thanks - but I need the bounding box rather than the centroid. I've updated the database so that now it contains both plots and subplots.

NewcombMaria, 2017-07-11:
We couldn't figure out how to do boundary geometries, just the centroids.  There's still a need to do some additional QC checking in the East/West (gantry Y axis) direction.  It looked pretty good to us when layered with UAV mosaic but could be off slightly E/W (less than 10 cm off probably, if at all).

NewcombMaria, 2017-07-14:
@dlebauer Andy and Rick completed the necessary QC check on the shape files and found that the plot boundaries needed to be shifted on the gantry Y axis 3.75 cm to the East.  I've saved the new post-QC shape files to the google directory https://drive.google.com/drive/folders/0B1OR5-zeXagneWVJNVk1S05kQlk   (to avoid confusion I removed the incorrect versions). 
Corrected whole-plot (2-row plots) and subplot post-QC files were generated (with the 3.75 cm shift to the East).  The new polygons will more accurately fit the actual plots if you are able to either add the shift into the geometries you already created, or start from the post-QC shape files and re-create the polygon geometries.  Let me know if you have questions about the shift.

dlebauer, 2017-07-14:
@NewcombMaria Thanks! I've updated the plot boundaries in the database

NewcombMaria, 2017-10-25:
Looks like this can be closed.  Plot polygon bounding box files have been quality-checked and remaining tasks are covered in issues #196 and #197

closed 2017-10-25 by NewcombMaria