terraref/reference-data #127: 3D data to delete

smarshall-bmr, 2017-05-12:
The laser didn't come on during last night's run. Please delete 3D data starting with 2017-05-11_21-24-53-664 until 2017-05-12_07-20-06-275.

dlebauer, 2017-05-12:
done

```sh
rm -rf /projects/arpae/terraref/sites/ua-mac/raw_data/scanner3DTop/2017-05-11/2017-05-11__2*
rm -rf /projects/arpae/terraref/sites/ua-mac/raw_data/scanner3DTop/2017-05-12
```

dlebauer, 2017-05-12:
@max-zilla could you please delete these records from Clowder?

dlebauer, 2017-05-18:
```sh
curl -X POST -d '{"extractor":"terra.environmental.energyfarm_datparser"}' -H "Content-Type: application/json" https://terraref.ncsa.illinois.edu/clowder/api/files/58f63f404f0c5bee63b41e06/extractions?key=KEY

curl -X POST -d '{"extractor":"terra.environmental.energyfarm_datparser"}' -H "Content-Type: application/json" https://terraref.ncsa.illinois.edu/clowder/api/files/58f63f404f0c5bee63b41e00/extractions?key=KEY
curl -X POST -d '{"extractor":"terra.environmental.energyfarm_datparser"}' -H "Content-Type: application/erraref.ncsa.illinois.edu/clowder/api/files/58f63f404f0c5bee63b41dfb/extractions?key=KEY
 
```

Status OK

max-zilla, 2017-05-18:
@dlebauer i havent had time to delete these yet. will try to write something up before next week.

closed 2018-01-24 by dlebauer