terraref/reference-data #129: Duplicate datapoints in Geostreams JSON from Energy Farm

infotroph, 2017-05-23:
### Description
I was wondering why the download kept failing, then realized it was 500 MB instead of ~20. Here's a smaller demo using PEcAn's Geostreams interface:

```
library("PEcAn.data.atmosphere")
library("dplyr")

ne = download.Geostreams(
  outfolder="~/gstest",
  sitename="UIUC Energy Farm - NE",
  start_date="2016-03-01",
  end_date="2016-03-31")
# 2017-05-23 15:07:59 INFO   [download.Geostreams] : 
#    https://terraref.ncsa.illinois.edu/clowder/api/geostreams/datapoints?sensor_id=411&since=2016-03-01&until=2016-03-31 

se = download.Geostreams(
  outfolder="~/gstest",
  sitename="UIUC Energy Farm - SE",
  start_date="2016-03-01",end_date="2016-03-31")
# 2017-05-23 15:09:00 INFO   [download.Geostreams] : 
#    https://terraref.ncsa.illinois.edu/clowder/api/geostreams/datapoints?sensor_id=410&since=2016-03-01&until=2016-03-31 

cen = download.Geostreams(
  outfolder="~/gstest",
  sitename="UIUC Energy Farm - CEN",
  start_date="2016-03-01",
  end_date="2016-03-31")
# 2017-05-23 15:09:16 INFO   [download.Geostreams] : 
#    https://terraref.ncsa.illinois.edu/clowder/api/geostreams/datapoints?sensor_id=412&since=2016-03-01&until=2016-03-31 

file.info(list.files("~/gstest", full.names=T))
#                                                                                             size
# /home/a-m/black11/gstest/Clowder.UIUC Energy Farm - CEN.2016-03-01.2016-03-31.2016.json 53357824
# /home/a-m/black11/gstest/Clowder.UIUC Energy Farm - NE.2016-03-01.2016-03-31.2016.json   1694881
# /home/a-m/black11/gstest/Clowder.UIUC Energy Farm - SE.2016-03-01.2016-03-31.2016.json   3377084
# ...snipped...

ne_parsed= jsonlite::read_json(ne$file, simplifyVector=TRUE, flatten=TRUE)
se_parsed= jsonlite::read_json(se$file, simplifyVector=TRUE, flatten=TRUE)
cen_parsed= jsonlite::read_json(cen$file, simplifyVector=TRUE, flatten=TRUE)

# Does CEN have more timepoints, or are some timepoints duplicated?
dups <- function(data){
  data %>% summarize(
    n=n(), 
    n_id=n_distinct(id),
    n_start=n_distinct(start_time),
    n_end=n_distinct(end_time))
}

dups(ne_parsed$data)
#      n n_id n_start n_end
# 1 2876 2876    2876  2876
dups(se_parsed$data)
#      n n_id n_start n_end
# 1 5752 5752    2876  2876
dups(cen_parsed$data)
#       n  n_id n_start n_end
# 1 94908 94908    2876  2876
# ==> SE has two of each timepoint, and CEN has 33(!) of each timepoint.

# Is the whole timepoint duplicated, or only some parameters?
varies = function(x) length(unique(x)) > 1
varies_within_time = function(data){
  (data 
  %>% group_by(start_time)
  %>% summarize_each(funs(varies))
  %>% summarize_each(funs(any), -start_time))
}

varies_within_time(ne_parsed$data) %>%  unlist() %>% .[.]
# named logical(0)
varies_within_time(se_parsed$data) %>%  unlist() %>% .[.]
#      id created 
#    TRUE    TRUE 
varies_within_time(cen_parsed$data) %>%  unlist() %>% .[.]
#      id created 
#   TRUE    TRUE 
# ==> in SE and CEN, each row has its own ID and the time created varies, but the met data are duplicated within each timepoint.
```

dlebauer, 2017-05-23:
Yep. Looks like the ratio of NE:SE:CEN records at each time step is 1:2:33

@robkooper &/or @max-zilla how to delete duplicates?

robkooper, 2017-05-24:
Lets make sure we also figure out how duplicates get into the database, so we don't have to do this all the time.

You can call `DELETE  /api/geostreams/datapoints/:id` agains clowder

max-zilla, 2017-05-24:
I think this is a result of the extractors evolving over time and being re-run, and not having a way to delete/recreate datapoints like we have for metadata so when new versions would get run it would append datapoints that already existed.

in the case of the energy farm data, i wonder if we just delete all datapoints from those streams, delete the metadata tracking "last read line" and just re-run them from scratch? should only take an hour or so for all 3...

dlebauer, 2017-05-24:
If that is the easiest way please do. Could you post the code? Might make
it easier to farm out the task in the future. I assume it won't require
100k delete statements ??

max-zilla, 2017-05-25:
@dlebauer actually quickest way, I think, is:

1. go to https://terraref.ncsa.illinois.edu/clowder/geostreams/sensors and delete the Energy Farm streams:
```
412 UIUC Energy Farm - CEN
411	UIUC Energy Farm - NE
410	UIUC Energy Farm - SE
```
2. go to the Energy Farm dataset: https://terraref.ncsa.illinois.edu/clowder/datasets/58f63f334f0c5bee63b41d56

3. for each of the three Avg15 files....
```
SE - https://terraref.ncsa.illinois.edu/clowder/files/58f63f404f0c5bee63b41e06
NE - https://terraref.ncsa.illinois.edu/clowder/files/58f63f404f0c5bee63b41e00
CEN - https://terraref.ncsa.illinois.edu/clowder/files/58f63f404f0c5bee63b41dfb
```
...delete the metadata from "terra.environmental.energyfarm_datparser" tracking datapoints created and last line read.

4. for those 3 files, Submit File for Extraction -> terra.environmental.energyfarm_datparser (with no parameters)

I will do these steps now.

max-zilla, 2017-05-25:
This process is done. Please close if the counts look better now.

rachelshekar, 2017-05-25:
@infotroph - please check data.

dlebauer, 2017-05-25:
It works. Thanks!

closed 2017-05-25 by dlebauer

dlebauer, 2017-05-27:
I checked data. It looks good

```r

ne_parsed= jsonlite::read_json(ne$file, simplifyVector=TRUE, flatten=TRUE)
se_parsed= jsonlite::read_json(se$file, simplifyVector=TRUE, flatten=TRUE)
cen_parsed= jsonlite::read_json(cen$file, simplifyVector=TRUE, flatten=TRUE)

dim(cen_parsed$data)

dim(se_parsed$data)

dim(ne_parsed$data)

met <- rbind(cen_parsed$data, se_parsed$data, ne_parsed$data)
met %>% group_by(start_time, sensor_name) %>% summarize(n = n()) %>% tidyr::spread(sensor_name, n)

```