terraref/reference-data #132: Add information about calibration targets

dlebauer, 2017-05-24:
For calibration targets used in terraref/computing-pipeline#281 we have

* the calibration certificate for the 75% panel 
  * response data extracted to table is in [this spreadsheet](https://docs.google.com/spreadsheets/d/1zSguW91E3KIgzbRvWtCvn-nXJz5R8nxbVg_KQ9TBb7Q/edit?usp=sharing):  
* the ID tag of the 20% panel. 
   * I've sent a request to Labsphere for the original calibration certificate. 

In addition, @solmazhajmohammadi measured each target with a spectrometer. @solmazhajmohammadi could you please add new folders for each target + spectra as a spreadsheet or text file to the Target Calibration Datasheets folder https://drive.google.com/drive/u/0/folders/0By_PDCcY5g2JS3BWX2MydTJhZDQ

Thanks

### Completion criteria
 
* [ ] name and record metadata for each target
* [ ] add these files targets metadata 
* [ ] add rsr curves for each target to google docs
* [ ] make rsr curves available in pipeline as clowder json objects in standard format

solmazhajmohammadi, 2017-05-24:
I uploaded the data. The output is sed file, you can open it as a text. It comes with information about the sensor as well. You can see 7 columns:
1st column: wavelength
2nd: Raw counts of reference in this case the 99% target
3d: Irradiance of reference (99% spectralon target)
4th: Raw counts of target (20 / 48 / 75 / 95 or 99% target)
5th: Irradiance of target (20 / 48 / 75 / 95 or 99% target)
6th: -log of reflectance, reflectance is internally calculated based on the target and reference 
7th: reflectance in %

czender, 2017-05-25:
I looked at the S_48% data (because it was first, and close to the high end of plant reflectance). Most of the reflectances were near 99% up to ~1.5 um. Am I misreading the labeling?

solmazhajmohammadi, 2017-05-25:
Updated...
please check the new files

dlebauer, 2017-07-27:
@craig-willis see here: https://github.com/pi4-uiuc/Team_1

craig-willis, 2017-08-23:
@dlebauer I've started organizing these by target.  I have the provided information for the 20% and 75% targets. I can extract some details from the paper for the 48% target/tarp.  Is the 99% target the existing SphereOptics (  https://terraref.ncsa.illinois.edu/clowder/files/5873a9544f0cad7d8131b36a)?  In this case the certificate describes a 99% panel, but the data is for a 95% (this is from an old git issue). I don't see any additional information about a 95% target.

rachelshekar, 2017-09-07:
@smarshall-bmr  - can you supply information about the 99% panel?