terraref/reference-data #134: Wrong ordering in Cropcircle data

solmazhajmohammadi, 2017-06-02:
Order of the indices in cropcircle data is wrong. 
Currently data is recorded with following order:
﻿<LemnaTecData> 

    <Entry name="Red Edge" value=" 0.052" unit="-" /> 
    <Entry name="NIR" value=" 0.111" unit="-" />
    <Entry name="RED" value="27.953" unit="-" />
    <Entry name="NDRE" value=" 0.115" unit="-" />
    <Entry name="NDVI" value=" 0.041" unit="-" />
    <Entry name="Distance to canopy" value=" 23.599" unit="mm" />
    <Entry name="est LAI" value=" 26.172" unit="-" />
    <Entry name="est CCC" value=" 20.948&#xD;" unit="-" />
</LemnaTecData>

However, the order should be changed to: 

﻿<LemnaTecData> 

    <Entry name="NDRE" value=" 0.052" unit="-" /> 
    <Entry name="NDVI" value=" 0.111" unit="-" />
    <Entry name=""Distance to canopy" value="27.953" unit="cm" />
    <Entry name="est LAI" value=" 0.115" unit="-" />
    <Entry name="est CCC" value=" 0.041" unit="-" />
    <Entry name="Red Edge" value=" 23.599" unit="-" />
    <Entry name="est NIR" value=" 26.172" unit="-" />
    <Entry name="est RED" value=" 20.948;" unit="-" />

</LemnaTecData>

Please note, unit of "Distance to canopy"  has been changed to cm.
The update has not been applied yet, I'll post the software update date here.

dlebauer, 2017-06-02:
What are the steps to resolve this and who is responsible?

dlebauer, 2017-06-08:
@solmazhajmohammadi please write out algorithm &/or bash script that will perform this fix. We can go through and find/replace the incorrect strings

max-zilla, 2017-06-29:
may need to update gantry software.

rachelshekar, 2017-07-26:
@solmazhajmohammadi please update this issue

max-zilla, 2017-08-10:
@solmazhajmohammadi any updates on this one?

rachelshekar, 2017-08-10:
Marcus has not updated the software yet.  We will need to wait until after code freeze for next release for this fix.  @solmazhajmohammadi will follow up.

max-zilla, 2017-08-10:
@solmazhajmohammadi mentioned that Markus has not fixed this yet.

@craig-willis this is just a labeling swap, lets take a quick look and see if we can address this in our metadata cleaning.

max-zilla, 2017-08-10:
OLD:
``
<Entry name="Red Edge" value=" 0.052" unit="-" /> 
<Entry name="NIR" value=" 0.111" unit="-" />
<Entry name="RED" value="27.953" unit="-" />
<Entry name="NDRE" value=" 0.115" unit="-" />
<Entry name="NDVI" value=" 0.041" unit="-" />
<Entry name="Distance to canopy" value=" 23.599" unit="mm" />
<Entry name="est LAI" value=" 26.172" unit="-" />
<Entry name="est CCC" value=" 20.948&#xD;" unit="-" />
```
NEW:
```
<Entry name="NDRE" value=" 0.052" unit="-" /> 
<Entry name="NDVI" value=" 0.111" unit="-" />
<Entry name="Distance to canopy" value="27.953" unit="cm" />
<Entry name="est LAI" value=" 0.115" unit="-" />
<Entry name="est CCC" value=" 0.041" unit="-" />
<Entry name="Red Edge" value=" 23.599" unit="-" />
<Entry name="NIR" value=" 26.172" unit="-" />
<Entry name="RED" value=" 20.948" unit="-" />
```

So 2 steps. 
Name swaps:
```
Red Edge -> NDRE
NIR -> NDVI
RED -> Distance to canopy
NDRE -> est LAI
NDVI -> est CCC
Distance to canopy -> Red Edge
est LAI -> NIR
est CCC -> RED
```
Value fixes:
```
est CCC -> remove extraneous characters (&#xD;)
```

dlebauer, 2020-06-05:
@max-zilla has this issue been resolved? Have fixes been back-propagated to historical data?