terraref/reference-data #136: Missing flirIrCamera filesbased on QA/QC cache data checks

NewcombMaria, 2017-06-07:
@dlebauer we had a meeting today to discuss the missing Flir files in the QA/QC data check reports.  @smarshall-bmr and @rjstrand provided some potential explanations, but we need more info on what info is used as criteria for a data set that is or is not good.  I've selected a day to demonstrate the problem.  Copied below is the data check from 2017-06-04. Stuart said it's possible that some of the images are shaded by the arms of the gantry and are low file size and are not considered 'good'.  What are the criteria for a good data set?  Could someone look the data directories from June 4 to see when/where the missing 68 files are during the scan?

9028 of 9028 Data Sets are good for co2Sensor
9028 of 9028 Data Sets are good for cropCircle
8033 of 8101 Data Sets are good for flirIrCamera
9028 of 9028 Data Sets are good for ndviSensor
9028 of 9028 Data Sets are good for priSensor
8016 of 8016 Data Sets are good for stereoTop
Note: This is a daily report of files received by the gantry-cache from the prior day's data sets.  There is a chance data generated on the prior day may still be transfering to the gantry-cache and thus is not accounted for in this report.  Good data sets are ones that contain the appriate number of files from each file type for that respective sensor.

dlebauer, 2017-06-07:
@jdmaloney can you check what the issue is? the badds folder says:

```
flirIrCamera:2017-06-05__12-04-49-084
json: 1 of 1
flirIrCamera:2017-06-05__12-04-49-084
json: 1 of 1
flirIrCamera:2017-06-05__12-04-49-084
json: 1 of 1
flirIrCamera:2017-06-05__12-04-49-084
json: 1 of 1
flirIrCamera:2017-06-05__12-04-49-084
json: 1 of 1
```

but it isn't clear what this means, or why the same line is duplicated many times. I checked the folder and it appears that - like other folders - it contains both a .bin and a .json file

jdmaloney, 2017-06-08:
@dlebauer @NewcombMaria Looking into this here tonight to see what is happening.  Based on the output David provided here above, this could be an issue with the checking code not handling something right it should be.  Will update this ticket here soon with what I find tonight and/or first thing tomorrow morning.  Thanks for bringing this to my attention.

jdmaloney, 2017-06-08:
@dlebauer @NewcombMaria 
Two part problem:
1. My script that generates that email had a bad syntax error and thus led to that repetitious directory (that was not a bad dataset) being put into the bad dataset text file.  Fixing up that code here now so it reports accurate and more useful output.  
2. The 8033 of 8101 datasets being good was a correct assessment of the output.  The data from this sensors is very consistent in size (I'm **not** checking size to generate above email, just the presence of the file suffices the check) but since it is consistent, I was able to see what directories were not of the normal 608KB per data set and was led to the attached 68 directories that each only had 4KB of data (just the .json file).  So it seems some datasets between 11am and 2pm were missing .bin files.

I'm sure having the correct list of problem datasets to look for will help you guys greatly :)  Sorry for the error in the script.  Again thanks for brining this to my attention, let me know if you need further investigation on my side for these.

[2017_06_04_Bad_FlirIr_Datasets.docx](https://github.com/terraref/reference-data/files/1059688/2017_06_04_Bad_FlirIr_Datasets.docx)

NewcombMaria, 2017-06-08:
@jdmaloney thanks for your helpful responses.  @smarshall-bmr and @rjstrand do you think the issue with the directories that only have the .json metadata file is a problem with the flir server???  The Flir came up short again from yesterday's June 7 data check report:
21177 of 21177 Data Sets are good for co2Sensor
16435 of 16435 Data Sets are good for cropCircle
**11992 of 12070 Data Sets are good for flirIrCamera**
21177 of 21177 Data Sets are good for ndviSensor
21177 of 21177 Data Sets are good for priSensor

NewcombMaria, 2017-06-09:
@smarshall-bmr and @rjstrand the report from yesterday (June 8) doesn't list stereoTop and also doesn't look good for flirIRCamera files.  Any ideas what happened to Stereo-vis images:

One Day QA/QC Gantry-Cache Data Check 2017-06-08
9063 of 9063 Data Sets are good for co2Sensor
9063 of 9063 Data Sets are good for cropCircle
601 of 8144 Data Sets are good for flirIrCamera
9063 of 9063 Data Sets are good for ndviSensor
9063 of 9063 Data Sets are good for priSensor
Note: This is a daily report of files received by the gantry-cache from the prior day's data sets.  There is a chance data generated on the prior day may still be transfering to the gantry-cache and thus is not accounted for in this report.  Good data sets are ones that contain the appriate number of files from each file type for that respective sensor.

NewcombMaria, 2017-06-12:
Stuart and Bob worked with Markus last week June 8 to diagnose the problem related to missing Flir files, something to do with network utilization and link speed.  The June 9 report that I received today looks like the problem is fixed.  Thanks @smarshall-bmr and @rjstrand!  This issue can be closed.  We'll need to keep an eye on the Flir directories since we don't know why the problem started.  
Two Day QA/QC Check 2017-06-09
9080 of 9080 Data Sets are good for co2Sensor
9080 of 9080 Data Sets are good for cropCircle
8160 of 8160 Data Sets are good for flirIrCamera
9080 of 9080 Data Sets are good for ndviSensor
9080 of 9080 Data Sets are good for priSensor
8060 of 8060 Data Sets are good for stereoTop

closed 2017-06-12 by NewcombMaria