terraref/reference-data #137: Define Danforth experiments in betydb experiments table

dlebauer, 2017-06-07:
For each experiment please 

* [ ] Create new experiment https://terraref.ncsa.illinois.edu/bety/experiments/new
   * [ ] Scott or I can link these to the Danforth site (can't be done in the web interface)
* Create new treatment(s) https://terraref.ncsa.illinois.edu/bety/treatments/new
* Create managements https://terraref.ncsa.illinois.edu/bety/managements/new and 
   * these should include planting date, harvest date, irrigation, fertilization
   * these should be automatically linked to treatments if you edit treatment --> new management
   * if you need help please let me know (may need to use new units of mass/plant rather than mass/area

If any of this becomes difficult, please put the information in a spreadsheet and I can help.

nfahlgren, 2017-06-15:
@dlebauer I will have to ask @nshakoor for this information.

dlebauer, 2017-12-01:
@nfahlgren and @nshakoor just FYI still waiting on this.