terraref/reference-data #138: Protocol for quality metrics and uncertainty estimates to datasets

dlebauer, 2017-06-08:
Based on previous discussions:

* https://github.com/terraref/reference-data/issues/130
* https://github.com/terraref/reference-data/issues/77

Write a protocol for capturing uncertainty estimates 

### Completion Criteria

* [ ] Capture discussions from related issues
* [ ] Draft protocol in google docs https://docs.google.com/document/d/1hWqkowvopYqGkeckSWg-_JzN3-DIS36rCBCFI09Sqyk/edit#
* [ ] Create new issues for annotating each dataset w/ quality and uncertainty information

rachelshekar, 2017-07-20:
@rmgarnett - can you work on this?  We need to know what should be included in the metadata by August 1.

rmgarnett, 2017-07-20:
Yes, sorry, I had written a follow-up email but should have posted here.  I was seeking clarification on the expectations for a protocol; I know @dlebauer has brought up this issue multiple times.

rmgarnett, 2017-07-23:
I have edited the document boiling it down to its essence. Please let me know if it looks okay.

rmgarnett, 2017-07-23:
@rachelshekar Can you clarify your metadata comment?  Do you want to store the predictions themselves in the metadata?