terraref/reference-data #145: Deploy Shiny App

nickheyek, 2017-06-12:


dlebauer, 2017-07-06:
Can prob. start w/ rocker/shiny docker container: https://hub.docker.com/r/rocker/shiny/

dlebauer, 2017-08-23:
See Dockerfile started in https://github.com/terraref/traitvis-webapp and notes on use in the [repository's README](https://github.com/terraref/traitvis-webapp#docker-notes)

dlebauer, 2017-08-31:
Many of the dependencies will be in this container: https://github.com/rocker-org/geospatial/blob/master/3.3.1/Dockerfile

* revert to exposing environment variables as in that file
  * you can develop using the ssh tunnel, and the only thing that may need to change for deployment is the betydb host
* add the packages listed in the README

dlebauer, 2017-10-05:
deployed here: http://traitvis.terraref.ndslabs.org/

closed 2017-10-05 by dlebauer