terraref/reference-data #147: Sorghum plots with mixed cultivars hand-planted after poor emergence

NewcombMaria, 2017-06-14:
Sorghum season2 had 16 plots with poor emergence that were hand-planted with a different cultivar to fill the gaps, leaving the original seedlings that did emerge.  We hand-planted with 700D-BMR, so the plots became mixtures of the original cultivar and 700D-BMR.  What's the best way to flag these plots in BETYdb?  The cultivar name for these plots (sites) could be changed to indicate the mixture, or the mixture could be described in a 'notes' column.  See the spreadsheet in google docs for the list of plots:  https://docs.google.com/spreadsheets/d/1PM6i0Kb7zyVsOe3JVnTdcDUbuDp2LQxz8Q-jOEtdq7M/edit#gid=1965472292

dlebauer, 2017-06-14:
I've updated the database with the following SQL insert / delete / update statements:

https://gist.github.com/dlebauer/64426018f7268406733d94b41756c47c

This can be closed except that it would be useful to get feedback from @gpmorrisksu, @rickw-ward,  and others:
This approach treats these plots as mutts by creating 'dummy' genotypes with names like "Mixed RIL XXYYZZ and 700D-BMR". As such, these data should not confound GWAS and similar analyses, but still allows us to process the sensor data to traits in the same way and leaves open the opportunity for some future dedicated researcher to understand the data. Does that make sense? 

There are other options, including either ignoring these plots completely or more accurately store information about the mixtures (e.g. as % or as a bounding box around the RIL seedlings of interest). Both of these approaches would be more work, and anything related to more accurately storing this information would be a lot more work that I suspect may not be justified in light of our other priorities and overall objectives.

rickw-ward, 2017-06-14:
The "mutt" approach looks ok to me.

dlebauer, 2017-06-15:
I've implemented the mutt approach and will close this. We can re-open later if we want to revisit w/ a more sophisticated approach.

closed 2017-06-15 by dlebauer