terraref/reference-data #151: Dark Reference Measurment for SWIR Sensor

solmazhajmohammadi, 2017-06-22:
As part of the calibration pipeline for Hyperspectral camera, we need to do similar dark measurement as we did for VNIR sensor [(Issue#88).](https://github.com/terraref/computing-pipeline/issues/88) 
The measurements need to be done during the night and with covered lens. Use rotating mirror and collect the dark reference for the exposures from 5-50ms(every 5ms).
Since the dark reference measurement done during the scan is noisy(because of reflection of sunlight into the sensor box), I suggest to disable the dark measurements in SWIR sensor metadata prior to each scan.

rachelshekar, 2017-08-10:
Dark measurement on SWIR is disabled because the measurements are noisy because there is light getting in. They are also temperature dependent so @smarshall-bmr might relocate the instrument or reduce the window size to reduce the noise.  
We need dark counts v. temp to correct the data.

dlebauer, 2017-08-10:
we need to take a series of dark measurements at different temperatures in the dark in order to create a temperature dependent dark reference calibration 

This feature will be implemented for the 2018 release. The assumptions / known sources of error should be documented in the data products.

sbrad77, 2018-06-18:
Has this series been slated to be taken? @dlebauer

sbrad77, 2018-07-06:
Communication Record:
On Jul 5, 2018, at 11:56 PM, LeBauer, David Shaner - (dlebauer) <dlebauer@email.arizona.edu> wrote:
Hi Vasit

There are a lot of open GitHub issues related to your work that need to be reviewed. 

Below is one example - can you please review this issue and provide feedback - is it still relevant? Should Jeffrey take these measurements?

More generally, how would you like to coordinate the issues that exist on GitHub and  future work more generally?

Thanks

David

sbrad77, 2018-07-06:
Communication Record:
Hi David, 
I will review these and act accordingly. For the near term, I plan to focus hyper calibration that is why did not pay attention much in other issues.

Thanks for letting me know.

Vasit