terraref/reference-data #152: Sorghum season1 cultivar identities for Della, Dale, M81E

NewcombMaria, 2017-06-28:
When I was uploading season1 data to BETYdb I used this spreadsheet https://docs.google.com/spreadsheets/d/1CPNQNJ7Pwh7ZC5EnOTWuZNptMTMTkMgy7a5AunCqeFM/edit#gid=527722434 for the complete list of sites/cultivars.  All of the cultivars were accepted during the data upload except DELLA (see plots 518-522, 540-544, and 555-560).  Apparently DELLA is entered as cultivar PI566819.  **This raises a question about cases when the same accession is kept in different programs over many years if the cultivar identity should reflect the program.**  

Dale also has a PI identity, but Dale was accepted as the cultivar name during the data upload.  For sorghum season1, there are 3 accessions that we received from Bill Rooney, which we referred to as Dale, DELLA, and M81E in the planting entry list, and these are the same accessions as the BAP lines that we received from Steve Kresovich that we referred to by their PI numbers in the BAP section of the field.  

I contacted Bill Rooney to ask if Dale, Della, and M81E from his program are the same as the BAP lines.  The email communication follows:  
_**"Pedigree:  Dale  (Seed source 13CS2221) Is Dale the same accession as the BAP line PI_651495 ?
Pedigree:  DELLA (Seed source 14PR716) Is DELLA the same accession as the BAP line PI_566819 ? 
Pedigree:  M81E (Seed source PR718) Is M81E the accession PI_653411​ ?

On all three of these lines, it is likely that they are similar but they may not be EXACTLY the same.  Specifically, I did not get this seed from GRIN.  This seed has been in our program for a long time and we have increased it over the years.  Consequently, there is the chance that there is some genetic shifting over time.  

From a phenotypic perspective, however, I don't it expect them to be different.  From a genetic perspective, there may be small differences that have accrued over time and could and would likely be detected in sequencing.**_" (from Bill Rooney) 

If DELLA is identified by PI number, regardless of which breeding/research program it came from, the Dale should probably also be identified by PI number.  Or we could concatenate the program (TAM or BAP which is maintained by Clemson).

NewcombMaria, 2017-06-30:
@dlebauer this issue overlaps some with issue #100 .  Over the next few days I'll get back to uploading 2016 field measurements (issue #61 ).  Sorghum season1 Ranges 33, 34, and 35 include plots with 'DELLA' and Ranges 39, 40, 41 include plots with 'Dale'.  These are entries from TAM but are the same PI accessions as BAP lines maintained by Clemson.  What's the best cultivar identity for these lines maintained by TAM.  See above comment for explanation from Dr Rooney.

rachelshekar, 2017-09-07:
@solmazhajmohammadi - can this issue be closed?

craig-willis, 2018-06-13:
Stale issue, closing. Open a new issue if work remains

closed 2018-06-13 by craig-willis