terraref/reference-data #158: Leaf Area Index in BETYdb has 2 variable names and 2 variable definitions

NewcombMaria, 2017-07-10:
@dlebauer   When I uploaded 2016 sorghum season1 in-season destructive sample biomass and leaf area estimates I found that an existing variable LAI is defined as 'Formula: 3.618f * ((2.5.0f * (R800 - R680)) / (R800 + 6.0f * R680 - 7.5.0f * R450 + 1.0f)) - 0.118f; Citation: Boegh et al. (2002)'.

This definition doesn't fit for our estimate of Leaf Area Index, calculated from leaf area measurements made using the LI-COR Area Meter.  For now I created a new variable 'leaf_area_index_AreaMeter', defined as 'Leaf area index estimated as the ratio of square meters leaf area to square meters land area, with leaf area measured using LI-COR Area Meter'.  Can the variablesfor Leaf Area Index be merged by using a more generic variable definition and then different methods for estimating LAI? 

This is related to issue #61

dlebauer, 2017-07-10:
@NewcombMaria I would suggest that we use the same variable named LAI for measurements that use hyperspectral, leaf area meter, or destructive harvest, and differentiate the measurements by the methods field. Does this sound reasonable to you?

NewcombMaria, 2017-07-10:
Yes @dlebauer makes sense to me.  The steps for this would be the following?

- Determine a generic, accurate definition for LAI and create a new variable LAI by that description
- I will write a method for LAI calculated using the leaf area meter on destructively sampled leaves
- Someone else will need to write the method(s) for measurements that use sensor-derived data 

I can do the first 2 tasks above, although I only know how to edit and not remove variables.  Will this work?  Thoughts?

NewcombMaria, 2017-10-03:
@dlebauer I've changed the issue name to generalize to any duplicate or overlapping variable/trait names.  

- this 2017 season 4 is the first season in Maricopa with stalk diameter data.  I looked at existing variable names and don't find 'stalk' but find 'stem_diameter' in cm, and also 'stem_width' in mm defined as 3rd internode from ground.  The use of 'stalk' fits that we are including the leaf sheath in the measurement and the true stem is within the leaf sheath.  Should I create another variable name ('stalk_diameter')?  Also we measured at different heights/positions from ground.  Should each height/position be a distinct variable name, or should the height/position be a covariate?  For one date early in the season we measured immediately below the most recently fully emerged leaf so the covariate would be the leaf internode number, for other dates the covariate could be height from ground. 

- for repeat measurements on marked plants, I've used variable names: 'stem_elongated_internodes_number' and 'plant_basal_tiller_number' for 2016 data, taken from sorghum crop ontology (http://www.cropontology.org/ontology/CO_324/Sorghum) because they seemed accurate and descriptive (in fact I'm counting internodes not leaf nodes).  I see existing alternative variable names in variables list:  'leaf_node_count' and 'tiller_count'.  I'd like to continue with 'stem_elongated_internodes_number' and 'plant_basal_tiller_number' for 2017 data.

dlebauer, 2018-12-13:
To answer a few questions ...

* yes, we can move the formula + reference for the LAI reflectance index to methods.
* I think it is okay to just rename the ‘stem_diameter’ as ‘stalk_diameter’. I am not aware of any measurements in which the leaves were removed before measuring diameter.
* for height of the diameter measurement we can add the height as a covariate - though it would be sufficient to define the height in the method as ‘3rd internode from the ground’.