terraref/reference-data #161: Change in the Offset of the sensors With Temperature

solmazhajmohammadi, 2017-07-13:
As it was observed in issue #70 , the distance between two 3D camera is changing with temperature. We are expecting to see the similar behaviour in the rest of the sensor's offset as well. Please set similar diurnal scan for the rest of the sensors as well. You can use the LemnaTec target for the rest of the sensors as well. 
The diurnal scan would give us  a function to model the offset change with temperature. We still need to measure the offsets for one temperature and set it as our control point.
Few notes:
- Place the target right under the stereo camera lens(one of them)
- Measure accurately the distance from target to the camera
- to make sure the measurements are accurate, measure the position of the target in the gantry coordinate system

dlebauer, 2017-07-13:
@solmazhajmohammadi 

* Do all of our current data products contain errors related to this? Do you have an estimate of the magnitude of the errors?
* Can you please make separate issue(s) / subtask(s) to cover the integration of these functions into the metadata cleaning algorithms?
* how will the measurements be done (with a ruler? laser scanner? caliper?) 
* Do you need the measurements to be done under the full range of temperatures observed that the sensors will be exposed to, or do you expect the functions to behave consistently across this range?

solmazhajmohammadi, 2017-07-13:
- Yes, we have about 1.4345 mm change per Centigrade change in top 3d scanner alignment. 
- Sure, first we need to find how much is this change. There is an issue related to this already created for the 3D scanner.
- Laser scanner. Distance from target to the stereo camera can be measured with gantry.
- It is a linear function.

Here is the measurements for the 3D scanner offset:
[image]

dlebauer, 2018-04-04:
@pless can you comment on how or where this could be incorporated into the pipeline? Do we have sufficient data to parameterize the correction? Do we have the values of temperature that we need?