terraref/reference-data #162: BETYdb bulk upload not accepting local_datetime field

NewcombMaria, 2017-07-15:
@dlebauer or @gsrohde I'm trying to upload 2016 leaf porometer measurements that have a time stamp for each individual measurement.  I created a field in the CSV file called "local_datetime", but the column is ignored during the upload.  Next I tried making "local_datetime" a covariate variable, but received the error message:
"Variable value can't be parsed as a number in these rows: 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11 ..."

Any suggestions for how upload the date/time? I have it formatted as 2016-06-15T11:48:41 
I have the csv file ready for upload if I can figure out the time stamp.

dlebauer, 2017-07-15:
> Assigned #162 <https://github.com/terraref/reference-data/issues/162> to

Can you please link to our attach the CSV?

NewcombMaria, 2017-07-16:
@dlebauer the CSV file is named 'Sorghum_season1_porometerdata_June.csv' in the google directory:
https://drive.google.com/drive/folders/0B1OR5-zeXagnWjJaaWRlSENsVms

NewcombMaria, 2017-07-16:
In betydb I defined the two variables:
leaf_stomatal_conductance 
ambient_temperature 
Let me know if you have questions.  Thanks!

gsrohde, 2017-07-17:
@NewcombMaria There are two upload methods, and there are slight differences in the CSV file format they require.  First, there's the Bulk Upload Wizard, which is what I think you've mostly been using.  It requires the heading for the date field to be just `date`.  But unfortunately, it doesn't (yet!) handle time-of-day values in the date, so the value must be in the YYYY-MM-DD format.

Second, there's the upload API.  It uses either the heading `local_datetime` (if the times given are the time at the site; this only works if _all_ sites in the file have been assigned a time zone; I think this is the case for the Maricopa sites); or one may use the heading `utc_datetime` if the time values are in UTC time.  I'll check your file and see if it could be uploaded using the API.

gsrohde, 2017-07-17:
@NewcombMaria I checked your file, and it is suitable for upload provided that an `access_level` column is added.  Let me know if you need help using the API or if you would rather I go ahead and upload it.

NewcombMaria, 2017-07-18:
Thanks @gsrohde .  So far I've only used the bulk upload wizard and not the API method.  It would be great if you go ahead and upload using API.  I wrote a Method description for this data-set.  Does the API upload allow you to connect the data with the Method during the upload?  The method name for this 2016 data-set is 'Leaf porometer measurements for stomatal conductance'.  See:  https://terraref.ncsa.illinois.edu/bety/methods/6000000016 
I'll need to write a different Method description in the future when I get to uploading 2017 field measurements.  Let me know how the upload goes.  Thanks very much!!

gsrohde, 2017-07-21:
@NewcombMaria I uploaded this set of data.  The new traits have id numbers 6001787823–6001788300.

The API does accept a `method` column, and the new traits are all connected with the method "Leaf porometer measurements for stomatal conductance".

I deleted the variable "local_datetime" that you created.  This actually interfered with the API working correctly.  It shouldn't, but I didn't make my code immune to name clashes as it should be.

NewcombMaria, 2017-07-22:
@gsrohde thanks very much!

gsrohde, 2017-07-24:
@NewcombMaria Can this be closed?  I think any remaining work is covered by https://github.com/PecanProject/bety/issues/524.

NewcombMaria, 2017-07-24:
Yes I'll go ahead and close it.

closed 2017-07-24 by NewcombMaria