terraref/reference-data #164: BETYdb - MAC Season 3 & 4 Traits Not Available

nickheyek, 2017-07-24:
### Description
There are no trait records in BETYdb with site_id attributes corresponding to associated sites of the MAC Season 3 and MAC Season 4 experiments.

dlebauer, 2017-07-24:
This is correct. Just to confirm

* @NewcombMaria have you uploaded hand-measured field data from 2017 yet?
* @ZongyangLi it looks like you need to update the canopy cover extractor to use the polygons in BETYdb, and to subset them based on associated experiment.

NewcombMaria, 2017-07-24:
I have not started to upload 2017 hand-measurement data yet.  I entered a couple Management activities for the current sorghum season, but have not uploaded any field/hand data-sets.  That's the next project, starting soon.

NewcombMaria, 2017-10-06:
I added the 'Management' activities for 2017 sorghum season 4.  Field measurement data sets are progressively being added and I'm updating completed data-sets in #174

dlebauer, 2017-10-18:
Looks like this issue has been resolved.

closed 2017-10-18 by dlebauer