terraref/reference-data #165: Traits that should be covariates

dlebauer, 2017-07-24:
We should be careful to separate traits (measurements of interest) from covariates (contextual data required to interpret the traits).

Covariates can be defined by adding records to the traits_covariates_associations table, and can be labeled as 'required' or 'optional'. 

Two data sets that appear to be covariates are 'ambient_temperature' and 'incident PAR'. Both are optional, because they are redundant with the met data that we have at this site. But they should be stored as covariates rather than as traits. 

I think that number_plants_destructive_sample, number_tillers_destructive_sample,  Minutes_after_first_Incident_PAR should also be considered a covariate. This 

### TODO

* [ ] identify any other data entered as traits that should be moved to the covariates table
* [ ] add these three variables to traits_covariate_associationis
* [ ] move existing records to the covariates table

NewcombMaria, 2017-07-24:
Yes agreed covariate designation is better for these data.  The name for the 'ambient_temperature' should probably be edited to something like 'microsite_ambient_temperature' because it differs from the met data. It's the air temp directly at canopy level and varies among plots depending on neighbors and exposure to wind and shading etc.  

The web bulk upload has a trick that took me a while to learn.  If I hit Enter while still in an entry field then the 'Type' designation drop-down field disappears and I'm not able to select 'covariate'.  Trying to edit the variable does not give me the chance to change from default 'plant trait' to 'covariate'.  I had to learn to click outside of entry boxes instead of hitting Enter.  If possible, it would be nice to allow the 'plant trait' versus 'covariate' selection to be edited using the web interface.

dlebauer, 2017-09-19:
@gsrohde could you please move the records that should be covariates to the covariates table? I believe that these will be associated with the appropriate traits by entity_id.

gsrohde, 2017-09-22:
@dlebauer 

It's not so simple as this.

Starting with release 4.15, we allow an entity column in a bulk upload file (see issue https://github.com/PecanProject/bety/issues/466).  Moreover, if the name in that column is the same as the name of an existing entity, that entity is reused instead of a new entity being created.  (See the release notes: https://github.com/PecanProject/bety/releases/tag/betydb_4.15)

I'm now thinking this "entity re-use" policy was not a good decision.  For example, in the data file for issue https://github.com/terraref/reference-data/issues/162, there are two rows with value "MAC Field Scanner Season 1 Field Plot 707 Sample 1" in the entity column.  These rows represent measurements made 9 days apart and yet they have been connected with the same entity.

dlebauer, 2017-09-22:
If they were made on the same plant or leaf etc that is correct. In some cases Maria tagged a leaf and returned to it repeatedly, and this approach would reflect that.

 if 'sample 1' refers to the first randomly sampled plant of that day that isn't the same as the first sample on the next day, then that is incorrect. In this case an arbitrary unique value for each sample on that day would suffice, as would spending the date.

dlebauer, 2017-09-23:
> Moreover, if the name in that column is the same as the name of an existing entity, that entity is reused instead of a new entity being created.

I think this feature is as intended. The key is that we need to make sure that the definition of an entity is clear to the end users.

gsrohde, 2017-09-23:
I guess I had been under the impression that the notion of an entity encompassed not only a particular leaf, plant, stand, etc., but also that it encompassed it at a particular point in time, so that one couldn't have two distinct values of a given trait or covariate variable for a given entity.

Can I at least assume that a trait or covariate variable should have a unique value for any given entity _at any particular time_?

I guess for now I'll charily trust that uploaders of data know what they're doing vis-à-vis entities and leave things—the way bulk upload works regarding entites, that is—as they are.  But it does seem to me there is some danger that someone uploading data will use a name for an entity that has already been used before to describe something totally unrelated and thus cause traits to be associated that shouldn't be.

Anyhow, getting back to the problem at hand, I assume that what I have to do is find "ambient_temperature" traits, and for each one, find another trait with matching `entity_id` _and_ `date` and convert the "ambient_temperature" trait to a covariate of the other "matching" trait.

dlebauer, 2017-09-23:
> entity encompassed not only a particular leaf, plant, stand, etc., but also that it encompassed it at a particular point in time

Sorry if this was not clear. An entity like a leaf can be revisited.

> Can I at least assume that a trait or covariate variable should have a unique value for any given entity at any particular time?

Yes, for an entity (like leaf 1) it can only have a single temperature at any given time.

>  there is some danger that someone uploading data will use a name for an entity

You are correct - every plot could have an entity called 'leaf a'. Thus far, we have been using the convention of combining the plot name + the leaf identifier. However, it is also implied (in the case of plants as opposed to animals) that one entity cannot be in two places at the same time so having an entity named 'leaf a' in two different plots is not logical. 

Uploading named entities is intended for cases where the user has explicitly kept track of a particular entity (as when they revisit the same leaf on multiple dates, or want to identify it in an image). By contrast, in the case where a researcher goes to the field, picks up a random leaf (e.g. 'sample 1') takes _n_ measurements, and never returns, it is important that those (and only those) _n_ measurements be linked to a common entity. At the same time, the consequences of incorrectly labeling an entity inside a plot are small. Where we are keeping track of entities, it is generally when we are keeping track of properties below the level of the plot, and the plot is the standard unit of replication. So the risk is really more that we are loosing the opportunity to further partition variance into multiple sources, for example by treating differences among entities as random-effects in a statistical model.

gsrohde, 2017-09-30:
@dlebauer, @NewcombMaria I need to know exactly which things should be covariates of which other things before I start converting them.

### ANALYSIS

---

#### ambient_temperature

Traits involving variable 'ambient_temperature' are all from the
following data set:

**file:** Sorghum_season1_porometerdata_June.csv
**date uploaded:** 2017-07-21 UTC
**uploaded by:** Scott Rohde
**upload method:** API

This file has two variable columns: `ambient_temperature` and
`leaf_stomatal_conductance`.

The introductory comment for this issue suggests that the 'ambient_temperature' values
should be covariates of the 'leaf_stomatal_conductance' values rather
that being traits in their own right.

The following two SQL statements insert the 'ambient_temperature'
values as covariates and then delete them from the traits table:

```sql
/* Traits corresponding to variable 'ambient_temperature' should be
   covariates of the traits having matching entity_id and date: */

INSERT INTO covariates(trait_id, variable_id, level)
    SELECT
        t.id,
        t2.variable_id,
        t2.mean
    FROM
        traits t
    JOIN
        traits t2
            ON t.entity_id = t2.entity_id
                AND t.date = t2.date
                AND t.variable_id != t2.variable_id
    JOIN variables v ON v.id = t2.variable_id
    WHERE
        v.name = 'ambient_temperature'
    AND t.created_at <= '2017-07-22' AND t.created_at >= '2017-07-21'
    AND t2.created_at <= '2017-07-22' AND t2.created_at >= '2017-07-21';

/* Now delete the 'ambient_temperature' traits: */

DELETE FROM traits
    WHERE variable_id = (SELECT id FROM variables WHERE name = 'ambient_temperature')
    AND created_at <= '2017-07-22' AND created_at >= '2017-07-21';

/* Add 'ambient_temperature' as an optional covariate of
   'leaf_stomatal_conductance': */

INSERT INTO trait_covariate_associations (trait_variable_id, covariate_variable_id, required)
    VALUES ((SELECT id FROM variables WHERE name = 'leaf_stomatal_conductance'),
            (SELECT id FROM variables WHERE name = 'ambient_temperature'),
            'f');
```
---
#### incident_PAR
Traits involving variable 'incident_PAR' are all from the following data set:

**file:** Sorghum_season1_groundlevelPAR_BAP_revised.csv
**date uploaded:** 2017-07-02 UTC (7/1/2017 local time)
**uploaded by**: Maria Newcomb
**upload method:** Bulk Upload Wizard

This file has three variable columns: `incident_PAR`, `ground_level_PAR`,
and `Minutes_after_first_Incident_PAR`.  Values of the first two are
present in each of the 288 data rows, but values of
`Minutes_after_first_Incident_PAR` are given for only 24 of these rows.

Should 'ground_level_PAR' be the trait variable and 'incident_PAR' be
a covariate of it?  Should 'Minutes_after_first_Incident_PAR' be
treated as a covariate of 'ground_level_PAR'?

Is it really true that 'Minutes_after_first_incident_PAR' is known for only 24 or the 288 rows, or should the given values apply to multiple rows?

These questions have to be decided before SQL statements are written.

---
#### number_tillers_destructive_sample and number_plants_destructive_sample
Traits involving variables 'number_plants_destructive_sample' and
'number_tillers_destructive_sample' are all from the following data
set:

**file:** Sorghum_season1_ inseason_destructivesamples.csv
**date uploaded:** 2017-07-10 UTC (7/9/2017 local time)
**uploaded by:** Maria Newcomb
**upload method:** Bulk Upload Wizard

This file has 14 variable columns:
```
fresh_weight_stems
leaf_area_index_AreaMeter
number_tillers_destructive_sample
aboveground_fresh_weight_1meter_destructive_sample
SLA
aboveground_fresh_weight_5plant_destructive_sample
dry_weight_stems
canopy_height
dry_weight_leaves
fresh_weight_leaves
biomass_dry_leaves
biomass_dry_stems
dry_matter_fraction
number_plants_destructive_sample
```
Two are singled out for conversion to covariates:
`number_tillers_destructive_sample` and
`number_plants_destructive_sample`.  Should these be covariates of each
of the other traits or only some of them?  Should any of the other
traits be converted to covariates?  These questions should be answered
before doing any conversions.

Note that the variable name "number_tillers_destructive_sample"
appears in the database twice, resulting in duplicate trait rows.
(See https://github.com/PecanProject/bety/issues/540.)  These duplicates will need to be deleted.

**General comment:** Because the trait-covariate relation is implemented 
as one-to-many rather than many-to-many, a covariate that applies to
more than one trait has to be repeated.  For example, consider a row
of a CSV upload file having 8 variable-name columns.  If all 8 are
considered trait variables, we will get 8 trait table rows and 0
covariate table rows for each CSV file row.  But if 4 are considered
trait variables and 4 are considered covariates of each of the 4
traits, then we will get 4 trait table rows and 16 covariate table
rows.

Probably if I were implementing covariates from scratch, I would store all measurements in a single table, and use virtual tables (views) to figure out and display which should be considered covariates of which others based on the entity, time of measurement, and name of variable (or perhaps a flag added to the variables table).

NewcombMaria, 2017-09-30:
Thanks @gsrohde .  
For the file: **Sorghum_season1_porometerdata_June.csv**  it is correct that the trait is 'leaf_stomatal_conductance' and the covariate is 'ambient_temperature'.  

For the file: **Sorghum_season1_groundlevelPAR_BAP_revised.csv**  the trait is 'ground_level_PAR' and one covariate is 'incident_PAR'.  The values for 'Minutes_after_first_Incident_PAR' was my adding time so that the Incident PAR from one reading to the next could be extrapolated.  Perhaps ideally the extrapolation should have been done before I uploaded the data-set.  Let me know if you or @dlebauer think I should do the extrapolation so that there is an estimated 'incident_PAR' covariate value for every 'ground_level_PAR' trait value. 

For the file: **Sorghum_season1_ inseason_destructivesamples.csv**  it gets confusing.  I'm not certain.  It's possible that this data-set also needs some revision to fit the database.

NewcombMaria, 2017-10-01:
@gsrohde thinking more about data-sets associated with **file: Sorghum_season1_porometerdata_June.csv** and **file: Sorghum_season1_ inseason_destructivesamples.csv**:  

I will revise the porometer csv file to fill in interpolated 'incident_PAR' covariate values for every 'ground_level_PAR' measurement.  I'll also add a column for time as 'local_datetime' for every 'incident_PAR'.  If the bulk upload still does not accept local_datetime (close issue #162  ) then I will need to ask you to upload by API.  The time-stamp or local_datetime upload will also be an issue for tractor data (issue #192 ) so let me know the status of time data upload using bulk upload.  

The dataset for inseason destructive biomass samples is complicated because 2 different methods were followed on different dates (plants were small and manageable on first data but were much larger on later date).  I separated variables by method for the upload, but now I understand it's better to have one variable name with the values categorized by the different methods.  The only true covariates as you suggested are numbers of plants and tillers that comprised the samples.  I'll revise this csv file and will need to edit some variable names.

dlebauer, 2017-10-02:
@NewcombMaria with respect to 'incident PAR' Are the values currently in the file extrapolated from one time point to the next? Although PAR seems to be increasing, the timeseries appears to be confounded by error, so the interpolations jump from one to the next (I haven't yet looked at the time series of this date). It also isn't clear how 'minutes after first incident par' would be used. At the very least, the method and its assumptions should be documented in the protocol.

[image]

We measure PAR above the gantry at very high frequency (~1/s or so) so it should be possible to find an incident par measurement for each time point.

So in the end, I would recommend that we just put the ground level par in the database and then add the incident par covariate from the environmental logger stream.

dlebauer, 2017-10-02:
@NewcombMaria to confirm - for stomatal conductance, where is ambient_temperature measured? is it above the canopy, inside the canopy, on the leaf, or ?

dlebauer, 2017-10-02:
@NewcombMaria for the fresh weight / dry weight you are correct that the variable name should be independent of the method. So for a 'destructive sample' the relevant information can go into the methods. But the number of plants / number of tillers in the destructive sample may not need to be a covariate. These could be stored as traits - e.g. stand density or tiller density with units /m2.

NewcombMaria, 2017-10-02:
@dlebauer regarding the csv file for ground-level and incident PAR, the incident PAR was only recorded in the field at the time-points recorded (columns O and P in the file), other values were interpolated.  There's a chance that the Ceptometer 'light bar' used in the field is calibrated differently and might be off from environmental logger data. 

The stomatal conductance data are the output of the Decagon porometer device.  My understanding is that the temperature is 'air' temp, but it's difficult to figure out from the manual  (pages 54-58): http://manuals.decagon.com/Manuals/10711_Leaf%20Porometer_Web.pdf 
 
Might be easiest to discuss these datasets by phone.

NewcombMaria, 2017-10-02:
@dlebauer as discussed please remove the data associated with the csv **file: Sorghum_season1_ inseason_destructivesamples.csv** from the database and I'll revise variable names and associated methods before re-uploading.  Thanks. 
**file: Sorghum_season1_ inseason_destructivesamples.csv
date uploaded: 2017-07-10 UTC (7/9/2017 local time)
uploaded by: Maria Newcomb**

dlebauer, 2017-10-02:
OK I've deleted the following 

```sql
select name, 
          CAST(traits.created_at AS DATE) as created_at_date, 
          count(*) as n 
     from traits join variables on traits.variable_id = variables.id 
     where CAST(traits.created_at AS DATE) = '2017-07-10' 
     group by name, created_at_date order by n desc;
```

| name                                               | created_at_date | n  |
|----------------------------------------------------|-----------------|----|
| number_tillers_destructive_sample                  | 2017-07-10      | 34 |
| biomass_dry_leaves                                 | 2017-07-10      | 17 |
| dry_weight_leaves                                  | 2017-07-10      | 17 |
| dry_weight_stems                                   | 2017-07-10      | 17 |
| biomass_dry_stems                                  | 2017-07-10      | 17 |
| leaf_area_index_AreaMeter                          | 2017-07-10      | 17 |
| number_plants_destructive_sample                   | 2017-07-10      | 17 |
| SLA                                                | 2017-07-10      | 17 |
| canopy_height                                      | 2017-07-10      | 17 |
| aboveground_fresh_weight_1meter_destructive_sample | 2017-07-10      | 8  |
| fresh_weight_stems                                 | 2017-07-10      | 8  |
| fresh_weight_leaves                                | 2017-07-10      | 8  |
| dry_matter_fraction                                | 2017-07-10      | 8  |
| aboveground_fresh_weight_5plant_destructive_sample | 2017-07-10      | 8  |

using 

```sql
DELETE FROM TRAITS WHERE CAST (traits.created_at AS DATE) = '2017-07-10';
```

> 210 rows deleted

gsrohde, 2017-10-02:
@dlebauer This would be a good time to remove the duplicate variable `number_tillers_destructive_sample` (either the one with id 6000000225 or 6000000226).  Let me know if you want me to do this.

To be really cautious, the following tables should all be checked for references:  
```
benchmarks
covariates
current_posteriors
formats_variables
likelihoods
posterior_samples
priors
traits
```
Each has a `variable_id` column.  Here's a case having foreign-key constraints in place would be really nice: we wouldn't have to worry about accidentally deleting a row that was still referred to.

dlebauer, 2017-10-02:
@gsrohde yes, please delete those records as well

I agree about the foreign key constraints; prior to our discussion today I did not realize that they weren't in place!

gsrohde, 2017-10-02:
@gsrohde<https://github.com/gsrohde> yes, please delete those records as well

Done

gsrohde, 2017-10-02:
> I agree about the foreign key constraints; prior to our discussion today I did not realize that they weren't in place!

Some of the foreign key constraints are in place: covariates.variable_id must refer to a valid variable.  The corresponding constraint for traits.variable_id is only in place on the terra databases (and maybe the mepp one).

gsrohde, 2017-10-03:
@dlebauer @NewcombMaria OK, I ran the SQL statements in https://github.com/terraref/reference-data/issues/165#issuecomment-333275839 so `ambient_temperature` is now a covariate of `leaf_stomatal_conductance`.

@dlebauer If you want to bother putting subtasks under the the second and third items of the to-do list (one for each variable that's becoming a covariate, perhaps), I'll check them off.

gsrohde, 2017-10-03:
@dlebauer By the way, I did a data dump of bety6's database.  The dump file is about an eighth the size of what it was in July.

dlebauer, 2017-10-03:
@gsrohde that is expected since we are purging lots of the automated datasets before rerunning the pipeline

gsrohde, 2017-10-03:
OK, I thought maybe that was it.  Just wanted to make sure.

NewcombMaria, 2017-10-27:
@gsrohde and @dlebauer I worked on the 2016 season1 in-season biomass and leaf area estimate data set that was based on destructive samples (data was deleted on 2017-10-02 as discussed in previous comment).  I revised the variables, excluding some and adding some, and all are trait variables this time (none are covariates).  Today I re-uploaded the revised data set, as two different data files for the two sample dates.  By re-uploading the revised data, two new tasks need to be completed: 

- [ ] The traits uploaded with this revised data set (11 trait variables in total - let me know if it's easier if I list the 11 variables) will need to be associated to the Method 'In-season leaf area and biomass estimates by destructive sampling' https://terraref.ncsa.illinois.edu/bety/methods/6000000015 

- [ ] Variables that were dropped can be removed from betydb (it looks like I can edit variables but not delete them from the web interface - is that correct?).  The 9 that can be removed include: 
   number_tillers_destructive_sample
   dry_weight_leaves 
   dry_weight_stems
   leaf_area_index_AreaMeter 
   number_plants_destructive_sample
   aboveground_fresh_weight_1meter_destructive_sample 
   fresh_weight_stems	
   fresh_weight_leaves 
   aboveground_fresh_weight_5plant_destructive_sample 

Let me know if more info is needed.  Thanks.