terraref/reference-data #167: BETYdb Season 2 Inaccurate/Missing Managements/Treatments Assocations

nickheyek, 2017-07-26:
### Description
In BETYdb, traits from season 2 have either no associated treatment, or treament 6000000014.
Treatment 6000000014, named 'control', is not associated with any management records that have a 'date' attribute within season 2.
Thus there are no management records in BETYdb corresponding to traits in Season 2.

Queries using dplyr BETYdb package:

```
season2_treatments <- tbl(bety_src, 'traits') %>% 
filter(date >= season2_start_date & date <= season2_end_date) %>% 
select(treatment_id) %>% 
collect() %>% unique() %>% na.omit() %>% as.numeric()

treatment_mgmts <- tbl(bety_src, 'managements_treatments') %>% 
filter(treatment_id %in% season2_treatments) %>% 
collect()

season_mgmts <- tbl(bety_src, 'managements') %>%
filter(id %in% treatment_mgmts[['management_id']]) %>%
filter(date >= season2_start_date & date <= season2_end_date) %>% 
collect()

nrow(season_mgmts)
[1] 0
```

NewcombMaria, 2017-07-27:
For season2 field measurements (hand measurements) using bulk upload process I selected 'control' for the treatment option because we did not implement any experimental treatments that involved management activities.  The whole field was treated as one management block.  @dlebauer is 'control' the best treatment description?  Another option could be something like 'RIL population' since the 'experiment' was designed for genotype/phenotype association studies using the SC56*Tx7000 RIL population.

NewcombMaria, 2017-08-03:
@rachelshekar and @dlebauer I'll need instructions on what needs to be done and how to make edits to treatments. For sorghum seasons 1 and 2 there were no experimental management treatments.  The whole field was managed as one block, and was managed for 'optimal' growing conditions.

dlebauer, 2017-08-10:
@NewcombMaria yes, 'control' is the best treatment description, as this typically refers to optimal growing conditions that a farmer would use.