terraref/reference-data #171: Update list of .nc files daily for thredds server

dlebauer, 2017-08-15:
currently the thredds server requires a list of .nc files that is parsed to an xml catalog used by the THREDDS server at https://terraref.ncsa.illinois.edu/thredds/catalog.html

However, the file has not been updated since July 17. This should be updated daily, prior to the generation of the xml file (should these be done in sequence in the same script?) 

Assigned to @jdmaloney; tagging @robkooper

jdmaloney, 2017-08-22:
@dlebauer @robkooper Cron job now in place to update this every day at 11:58pm (though I just did it once by hand).  Output for Rob kept in /gpfs/largeblockFS/projects/arpae/terraref/sites/ua-mac/Level_1/hyperspectral/nc_files

jdmaloney, 2017-08-24:
This is all working now on a daily basis

dlebauer, 2017-08-24:
I am not sure if anything changed, but now the directory tree in the UI provides no information about which files will be found at each of the links called 'project'. 

I suspect this may be a bug - did you change the script, and is there a way to replicate the directory structure on roger?

This is how it currently looks:

[image]

But it would be great if it looked more like

> | ua-mac
| | hyperspectral
| |  |-2017-01-30
| |  |  |-2017-01-30__14-49-10-193
| |  |  |-2017-01-30__15-53-50-679
| |  |  |-2017-01-30__15-02-21-574

robkooper, 2017-08-24:
fixed, format had changed of the file listing.

now is:

[image]

closed 2017-08-24 by robkooper