terraref/reference-data #172: Add texture variables to BETYdb

dlebauer, 2017-08-29:
Steps:

1. Identify citation or write a protocol on protocols.io
2. Add a single method for the extractor terraref.ncsa.illinois.edu/bety/methods/new
3. Add a variable for each phenotype terraref.ncsa.illinois.edu/bety/variables/new or provide a table with four columns: Name, Description, Units, Notes

dlebauer, 2017-09-19:
From @solmazhajmohammadi:

Here is more info on the texture analysis extractor:

-table.csv
This file contains the feature vectors per image or region of interest. Following fields are available:
- roi = label of region of interest [1..N]
- area = area of ROI [px]
- edges = count of pixels
- dgci.-0.1 .. dgci.2.9 = histogram bins ranging from DGCI values -.1 to 2.9;
[the values represent frequency counts]

- s.area: area size of region of interest (in pixels)
- s.perimeter: perimeter of shape (in pixels)
- s.radius.mean: mean radius of shape (in pixels)
- s.radius.sd: standard deviation of the mean radius (in pixels)
- s.radius.max: max radius (in pixels)
- s.radius.min: min radius (in pixels)

- m.cx: center of mass x (in pixels)
- m.cy: center of mass y (in pixels)
- m.majoraxis: elliptical fit major axis (in pixels)
- m.eccentricity: elliptical eccentricity defined by sqrt(1-minoraxis^2/majoraxis^2). Circle eccentricity is 0 and straight line eccentricity is 1.
- m.theta: object angle (in radians)

For DGCI I used:
https://www.researchgate.net/publication/297410506_Quantifying_turfgrass_color_using_digital_image_analysis
 

For the shape and moment features I used EBImage: 
https://bioconductor.org/packages/release/bioc/html/EBImage.html
https://rdrr.io/bioc/EBImage/man/computeFeatures.html

You can average the DGC indexed pixel values and compute a mean DGCI value.
If there is something else missing, please let me know.

dlebauer, 2017-09-19:
@solmazhajmohammadi it is still not clear:

* how are ROIs are defined? 
   * Could you provide an example of an image with the ROI outlined? 
   * what do the edges mean?
   * does the definition of an ROI differ before, during, and after canopy closure?

solmazhajmohammadi, 2017-09-19:
We are not using ROI(region of interest) in the current extractor. If the plants are masked out from the background, then we can use that mask/ROI to feed into this pipeline.
Edges are the number of pixels detected by sobel operator (edge detection)

dlebauer, 2017-09-20:
@solmazhajmohammadi I was confused by the use of the term 'texture analysis'. After reading up on it and looking through the code I realized that this extractor currently doesn't perform 'texture analysis', though it seems straightforward to do so by adding `(computeFeatures.texture(roi))` to https://github.com/terraref/extractors-stereo-rgb/blob/master/stereo-texture-analysis/gift.R#L132 (or replacing that line with `computeFeatures(roi)` to compute all of the properties that the package can output).

In the current state, I think that 'texture  It seems that the outputs are geometric properties, similar to the ones generated by @nfahlgren's PlantCV.

Do you want to update the extractor to include feature analysis?

solmazhajmohammadi, 2017-09-20:
@dlebauer 
If the ROI doesnt exist, it is creating a mask matrix which all the elements are one
https://github.com/terraref/extractors-stereo-rgb/blob/master/stereo-texture-analysis/gift.R#L118
then it would apply the computeFeatures(roi) anyway. 
Is there anything need to be modified that I'm missing?

max-zilla, 2017-09-21:
@solmazhajmohammadi is asking someone at LemnaTec, but for now we should be able to use the DGCI histogram values.

solmazhajmohammadi, 2017-09-22:
Detailed information on how the feature is calculated: 
https://rdrr.io/bioc/EBImage/man/computeFeatures.html
To be clear, see the attached image on how the radius is calculated for any arbitrary shape:
[image]
[image]
[image]
[image]

dlebauer, 2017-09-22:
@solmazhajmohammadi sorry, but the following is still unclear - 

1. why are some circles outside and others inside the object?
2. In the second image, why is the circle not large enough to touch the edges of the object?
3. In an image of a closed canopy like the one below, what circle(s) would be used to compute these metrics? Would the image be treated like a box as we discussed yesterday?

[image]

dlebauer, 2017-09-22:
@max-zilla For DGCI I've added the variable [DGCI](https://terraref.ncsa.illinois.edu/bety/variables/6000000259) and the method [EBImage](https://terraref.ncsa.illinois.edu/bety/methods/6000000023)

solmazhajmohammadi, 2017-09-22:
@dlebauer 
1. inside and outside circles refer to min and max radius.
2. that's only a made-up image to clarify the point, correct it should be slightly larger to touch the edges.
3. true, whole image would be considered as a rectangle

dlebauer, 2017-09-25:
I am compiling a list of features in this google spreadsheet https://docs.google.com/spreadsheets/d/1iVxnfl3jJ1B5wLOdewEGwSNQ0R9IXhl7vqmIt9EKUv8/edit#gid=0

kevin05jan, 2017-09-29:
Image texture is a metric to quantify the spatial distribution of color in an image. One method is to use the statistical approach, i.e. by counting edge pixels.  For example an RGB image is converted into an intensity image, threshold is applied, and from the resulting binary image the amount of edge pixels reflect a pattern in the color image.

Here I compute the dark green color index (DGCI) image as a way of normalising and highlighting contrasts between green leaves.  The counting of edge pixels is based on the entire image (default) or by using a set of region of interest.  By using a set of ROIs you can compare local features.

Hope this helps.

dlebauer, 2017-09-29:
Thanks Kevin 

* How do you use a set of ROIs? 
  * Can you provide an example? 
* Which metrics, if any, are confounded by the size and shape of the image itself? eg in the absence of ROIs does radius measure the image rather than the plant?

kevin05jan, 2017-09-29:
One command argument of the developed R script allows the input of a binary
image mask.  Each white blob in the image mask represent a unique ROI.  The
command line options are documented and forwarded to Solmaz:

Rscript gift.R -f gift-test.png -t -r roi.png ### this analysis the image
using a b/w image as ROI mask

Regarding the metrics, they are based on a ROI shape.  If an image mask is
absent, then the morphological features are computed for the entire input
image.  The key in my R script is the DGCI based texture analysis.

dlebauer, 2017-09-29:
@kevin05jan so to be clear, the script that you provided will generate many non-sensical values (radius, perimeter, etc) if there is no ROI mask, and you haven't provided a way of generating an ROI mask?

So ... of the features that your script generated, only dgci and edge are measurements of the plant being imaged rather than the image itself? And it would seem that edge should be normalized by area in order to normalize for the size of an image?

kevin05jan, 2017-10-09:
The tool aims to analyse image texture on the basis of DGCI.  In context of an image processing pipeline its job is not to process an image (image segmentation) but to analyse it.  

The normalisation should be based on the area of interest, hence the ROI image file input.

dlebauer, 2017-10-09:
OK. I don't believe we have an image segmentation pipeline that generates the required components for this extractor (and even if we did it would be too late to couple this for the Nov 2017 release). We will discuss implementation for the 2018 release; @kevin05jan will you be able to develop the pre-requisite image segmentation pipeline?

kevin05jan, 2017-10-10:
Just to be clear:-
- is there no extractor to perform VIS based image segmentation?
- are the extractors meant to be combined in a pipeline, similar as in a
BASH shell with pipes? E.g.
    $>  echo This is the year 2017 | perl -pe 's/[ ]+/\n/g' | egrep -v
"[0-9]+"

I can develop another extractor for vis based image segmentation if
required.

dlebauer, 2017-10-11:
Yes, similar in concept to bash pipes but different in implementation - each component needs to be coupled using a pyclowder script called an extractor.