terraref/reference-data #173: Changes to hyperspectral indices file

dlebauer, 2017-08-29:
For each index, there are two variables in the indices file, the average value and the full resolution index.

* remove all of the scalar variables that are the mean value of an index; only keep the raster (x,y) values
    * The average value is not useful, except perhaps as a diagnostic, since these files have not been subset to plots. When they are subset, this value should be insert into BETYdb, and optionally added as a metadata field (along with min and max?) but should _not_ need to be a separate variable in the Level 2 indices file. 
* remove the `_pxl` suffix from the high resolution variables
* change the metadata in these variables. 
   * Currently the full res. indices have a standard_name of 'surface_albedo'. But these variables should have metadata from that is currently queried from BETYdb and used to populate the scalar variables. 
* the following variables can be excluded from this Level 2 product.
   * image corners (e.g. lat_img_ne, x_img_ne etc)
   * [lon/lon/x/y]_reference_point
   * rfl_img, rfl_img_min, rfl_img_max, rfl_rfr_fct, wvl_db
   * solar_zenith_angle
* what is the variable `eta`?
* have the coordinate variables (x, y, latitude, longitude) appear first

related: https://github.com/terraref/reference-data/issues/83#issuecomment-323133962

hmb1, 2017-08-30:
@dlebauer 
@max-zilla 
Think I have ticked most of the above bullets.
The latest mods are now in master 
to test the nco script from the command line use the command
`ncap2 -v -O -S hyperspectral_indices_make.nco  <in.nc> <out.nc>`

is you want to add avg,min,max add the "flg_agg"
`ncap2 -v -O  -s '*flg_agg=1s;'  -S hyperspectral_indices_make.nco  <in.nc> <out.nc>`

Not sure what eta is (  I think Chris  added this one )

...Henry

hmb1, 2017-08-30:
@dlebauer 
@max-zilla 

to test the nco script from the command line use the command
`ncap2 -v -O -S hyperspectral_indices_make.nco  <in.nc> <out.nc>`

is you want to add avg,min,max add the "flg_agg"
`ncap2 -v -O  -s '*flg_agg=1s;'  -S hyperspectral_indices_make.nco  <in.nc> <out.nc>`

Not sure what eta is (  I think Chris  added this one )

...Henry

dlebauer, 2017-08-30:
Is there a sample somewhere?

hmb1, 2017-08-31:
see sample on ROGER:  **/tmp/dlebauer/2017-07-08__06-43-19-959_ind.nc**