terraref/reference-data #178: Review flir2tif extractor and output

craig-willis, 2017-09-06:
Review issue for the flir2tif extractor and ir_geotiff output.  

Completion criteria:
* [ ] Code review complete (includes extractor, terrautils, etc)
* [ ] Sample data output verified 
* [ ] Sample metadata verified
* [ ] Changes for 2017 release implemented
* [ ] Issue(s) for changes postponed until 2018 created 
* [ ] Extractor tagged/versioned
* [ ] Extractor deployed
* [ ] Final output verified/reviewed

max-zilla, 2017-09-07:
LINKS

**flir2tif output dataset example**
https://terraref.ncsa.illinois.edu/clowder/datasets/59b0ad984f0ca12ea0cdeff1
**raw dataset that was used to generate this output**
https://terraref.ncsa.illinois.edu/clowder/datasets/59b0ad974f0ca12ea0cdefd2
**collection with 11,395 other output datasets**
https://terraref.ncsa.illinois.edu/clowder/collection/59b04f7e4f0ca12ea0bed029

an output dataset includes PNG and GeoTIFF. 

outputs on ROGER are in /ua-mac/Level_1/ir_geotiff

max-zilla, 2017-09-07:
@ZongyangLi provided code updates here: https://github.com/terraref/reference-data/issues/182#issuecomment-327859158

...to attend to calibration. will run a separate day & fullfield using new parameters when @craig-willis returns tomorrow morning.

max-zilla, 2017-09-11:
https://terraref.ncsa.illinois.edu/clowder/datasets/59b2ec6c4f0ca12ea0d7a162
I ran a day with temperature calibration fixes. Here is one example dataset.

Full field result discussed in #182

ZongyangLi, 2017-10-04:
@max-zilla When I debugging the FLIR data, I think it is better to use the maximum scale for PNG files, I had updated two lines here: https://github.com/terraref/extractors-multispectral/blob/master/flir2tif/Get_FLIR.py#L291 
Please update the codes in your version.

dlebauer, 2017-10-05:
need to identify bad files as bad (all from Season 4)

closed 2018-06-07 by craig-willis