terraref/reference-data #179: Review envlog2netcdf extractor and output

craig-willis, 2017-09-06:
Review issue for the envlog2netcdf extractor and output.  

Completion criteria:
* [ ] Code review complete (includes extractor, terrautils, etc)
* [ ] Sample data output verified 
* [ ] Sample metadata verified
* [ ] Changes for 2017 release implemented
* [ ] Issue(s) for changes postponed until 2018 created 
* [ ] Extractor tagged/versioned
* [ ] Extractor deployed
* [ ] Final output verified/reviewed

See also: https://docs.google.com/spreadsheets/d/1vxhPPju6W9bDuUAcYFRD-KwihXyfuW6WECJbNmEBU84/edit#gid=0

dlebauer, 2017-09-07:
does this extractor generate netcdf AND push to geostreams?

max-zilla, 2017-09-07:
@dlebauer yes, that is correct.

LINKS

**envlog2netcdf output dataset examples**
- 06/15 https://terraref.ncsa.illinois.edu/clowder/datasets/59b18e864f0ca12ea0d231b6
- 07/30 https://terraref.ncsa.illinois.edu/clowder/datasets/59b18cde4f0ca12ea0d22be3
- 08/15 https://terraref.ncsa.illinois.edu/clowder/datasets/59b18db84f0ca12ea0d2310f
**raw dataset that was used to generate these outputs**
- 06/15 https://terraref.ncsa.illinois.edu/clowder/datasets/59b18ce54f0ca12ea0d22d8e
- 07/30 https://terraref.ncsa.illinois.edu/clowder/datasets/59b18cdb4f0ca12ea0d22afa
- 08/15 https://terraref.ncsa.illinois.edu/clowder/datasets/59b18cde4f0ca12ea0d22be0
**links to geostreams datapoint array outputs**
- map: https://terraref.ncsa.illinois.edu/clowder/geostreams#
- sensor_par https://terraref.ncsa.illinois.edu/clowder/api/geostreams/datapoints?key=Pb3AUSqnUw&stream_id=3207
- sensor_weather_station https://terraref.ncsa.illinois.edu/clowder/api/geostreams/datapoints?key=Pb3AUSqnUw&stream_id=3208
- sensor_spectrum https://terraref.ncsa.illinois.edu/clowder/api/geostreams/datapoints?key=Pb3AUSqnUw&stream_id=3209
- sensor_co2 https://terraref.ncsa.illinois.edu/clowder/api/geostreams/datapoints?key=Pb3AUSqnUw&stream_id=3210

an output dataset includes a netCDF file. 

outputs on ROGER are in /ua-mac/Level_1/envlog2netcdf

dlebauer, 2017-10-02:
* for the weather station, need to resolve the name of solar zenith angle "Solar zenith (or elevation) angle (under discussion right now)"
* need standard name for atmospheric temperature, zenith angle, brightness, wind speed
* need description for all variables except Relative Humidity

| |standard_name | description |
|:---|:---|:---|:---|
| Solar zenith angle  |solar_zenith_angle | angle between the line of sight to the sun and the local vertical.|
|~~Atmospheric~~Air Temperature | air_temperature | bulk temperature of the air, not the surface (skin) temperature. |
|Wind Speed  |wind_speed | magnitude of the wind velocity.|
| ~~brightness~~ Illuminance | illuminance | total luminous flux incident on a surface, per unit area

dlebauer, 2017-10-02:
Otherwise, data look reasonable here are some sample data (source code https://gist.github.com/dlebauer/5038f328b09d638b4f17ffa6f640868c):

[image]

[image]

[image]

dlebauer, 2017-10-02:
The only dataset that doesn't make intuitive sense is 'wavelengths' 

[image]

max-zilla, 2017-10-02:
@dlebauer I will need to add a small if statement to account for these name changes, the names come directly from the JSON file without modification so I'll add a find/replace here:
https://github.com/terraref/extractors-environmental/blob/master/envlog2netcdf/terra_envlog2netcdf.py#L70

dlebauer, 2017-10-02:
@max-zilla can you also remove the 'wavelength' data from geostreams?(This is used as a dimension for measurements by the downwelling radiometer, and these are excluded from the geostreams database.) 

Then, this extractor is ready to go (2018 changes listed here: https://github.com/terraref/reference-data/wiki/Changes-for-2018#environmental-logger)

dlebauer, 2017-10-02:
plot of a time series from the radiometer (representing the same time range as above (approx 2:15-3:30 on Aug 15, 2017)

[image]

max-zilla, 2017-10-03:
Removed wavelenght, added descriptions & names for the ones you mentioned.

Also updated this to append each new .json file to a single .nc file per day as per @hmb1 request.

max-zilla, 2017-10-03:
DEPLOYMENT
- [ ] purge all environment logger streams & datapoints from geostreams
- [ ] purge all files from /Level_1/envlog_netcdf
- [ ] submit EnvironmentLogger raw data root collection with my extraction script to run through this extractor & create new full-day .nc files and geostreams datapoints

(eventually, deploying in a priority rush today)
- [ ] purge all .nc files from raw_data sets in Clowder (old runs)
- [ ] purge all data from /Level_1/EnvironmentLogger

craig-willis, 2018-06-07:
Stale issue -- closing. Open a new issue if work remains.

closed 2018-06-07 by craig-willis