terraref/reference-data #180: Review datparser extractors and output

craig-willis, 2017-09-06:
Review issue for the irrigation, weather, and energy datparser extractors and output.  

Completion criteria:
* [ ] Code review complete (includes extractor, terrautils, etc)
* [ ] Sample data output verified 
* [ ] Sample metadata verified
* [ ] Changes for 2017 release implemented
* [ ] Issue(s) for changes postponed until 2018 created 
* [ ] Extractor tagged/versioned
* [ ] Extractor deployed
* [ ] Final output verified/reviewed

See also: https://docs.google.com/spreadsheets/d/1vxhPPju6W9bDuUAcYFRD-KwihXyfuW6WECJbNmEBU84/edit#gid=0

dlebauer, 2017-09-07:
Where can I find sample data / metadata?

max-zilla, 2017-09-07:
LINKS

**weather dataset examples**
- 06/15 https://terraref.ncsa.illinois.edu/clowder/datasets/59b193dc4f0ca12ea0d23467
- 07/30 https://terraref.ncsa.illinois.edu/clowder/datasets/59b193d54f0ca12ea0d232f1
- 08/15 https://terraref.ncsa.illinois.edu/clowder/datasets/59b193d64f0ca12ea0d23351

**irrigation dataset examples**  (only one CSV available per month)
- 07/02 https://terraref.ncsa.illinois.edu/clowder/datasets/59b197c94f0ca12ea0d238be
- 08/27 https://terraref.ncsa.illinois.edu/clowder/datasets/59b197c94f0ca12ea0d238ce
- 09/05 https://terraref.ncsa.illinois.edu/clowder/datasets/59b197c84f0ca12ea0d238b7

**links to geostreams datapoint array outputs**
- map: https://terraref.ncsa.illinois.edu/clowder/geostreams#
- weather observations https://terraref.ncsa.illinois.edu/clowder/api/geostreams/datapoints?stream_id=3211
- irrigation observations https://terraref.ncsa.illinois.edu/clowder/api/geostreams/datapoints?stream_id=3212

these extractors push to geostreams, with no output files created.

max-zilla, 2017-09-07:
ISSUE: Metadata cleanup
- irrigation correctly reports # datapoints created, but attaches to file rather than dataset. should attach to dataset
- weather incorrectly reports # datapoints created due to loop nesting issue.

Both issues have been addressed in code. Irrigation will write to dataset and weather will correctly count datapoints.

dlebauer, 2017-09-21:
I'll also check these in the PEcAn workflow

craig-willis, 2018-06-07:
Sale issue - closing. Open a new issue if work remains.

closed 2018-06-07 by craig-willis