terraref/reference-data #181: Review hand measured traits from KSU 2016

rachelshekar, 2017-09-07:
_PLEASE USE THIS TEMPLATE when submitting a new issue_

### Description (REQUIRED)

 
### Completion Criteria (REQUIRED/ESSENTIAL)

dlebauer, 2017-09-07:
@zhenbinHU do you know where the plot geospatial information is? It never made it into the database.

Also can you please check the following?

1. Flowering Time looks like an artifact: [image]
2. Is there really a 30mm wide leaf? [image]
3. Datasets currently on Drive but not in database: 
   * Sugar
   * leaf_length/width ratio
   * fresh biomass (also need moisture &/or dry biomsass)
   * ADF

zhenbinHU, 2017-09-07:
@dlebauer  for leaf width, the 300mm should be a typos. Do you from Ashland Bottoms KSU   2016 Season Range 17 Pass 13. If yes, it should be 85.66666667. Can you please change it?

For flowering time, for this kind of figure it should this pattern. I don't thinks it with problem.

dlebauer, 2017-09-07:
@zhenbinHU I've updated the geospatial information and the leaf width.

* should flowering time should have the units 'days after planting'?  
* there are 175 plots that have no geospatial information listed here: https://docs.google.com/spreadsheets/d/1APkedH70txKfvUknvgiqDTsdLWC2uriTj5dXIrHNY74/edit#gid=1833959272
* there are 501 plots with no associated cultivar listed here: https://docs.google.com/spreadsheets/d/1APkedH70txKfvUknvgiqDTsdLWC2uriTj5dXIrHNY74/edit#gid=71846873 
* canopy_height data that you uploaded on Aug 15 2017 is all associated with a single sitename 'KSU 2016 field season' ... but these records should be associated with individual plots. I have deleted these 195 records so that you can re-upload them with appropriate plot associations. The deleted traits are in 
[deleted_traits.txt](https://github.com/terraref/reference-data/files/1285579/deleted_traits.txt).

zhenbinHU, 2017-09-18:
@dlebauer Yes, the units of flowering time can be "days after planting"

zhenbinHU, 2017-09-21:
@dlebauer I have uploaded the canopy height data that you deleted. If you have question about the data please let me know.
For the 175 plots without geospatial information, these plots were fill, border or empty (GCP). I have asked Kevin the full field map. If you need I can send you the map file.
I will figure out the 501 plots that no associated cultivar soon.