terraref/reference-data #182: Review FLIR fullfield extractor and output

max-zilla, 2017-09-07:
Review issue for the fullfield extractor and output using flir thermal IR inputs.  

Completion criteria:
* [ ] Code review complete (includes extractor, any functions in terrautils, etc)
* [ ] Sample data output verified 
* [ ] Sample metadata verified
* [ ] Changes for 2017 release implemented
* [ ] Issue(s) for changes postponed until 2018 created 
* [ ] Extractor tagged/versioned
* [ ] Extractor deployed
* [ ] Final output verified/reviewed

max-zilla, 2017-09-07:
LINKS

**fullfield output dataset example**
https://terraref.ncsa.illinois.edu/clowder/datasets/59b0e2134f0ca12ea0ce678a

an output dataset includes thumbnail- and full-resolution GeoTIFFs. 

Ignore that this linked dataset has thousands of files, this is a glitch in extractor upload code. The top two .tif files are just duplicated endlessly so just look at the first ones listed.

outputs on ROGER are in /ua-mac/Level_1/fullfield.

dlebauer, 2017-09-07:
One of the full field IR stitched images looks like this:

[image]

With values ranging from 1221 to 1574. If this is temperature (K), values should mostly be in the range 275-300

dlebauer, 2017-09-07:
A lower priority issue is that these show up blank in the Clowder interface: 

[image]

ZongyangLi, 2017-09-07:
@max-zilla @dlebauer 
In the flir extractor code here: https://github.com/terraref/extractors-multispectral/blob/master/flir2tif/Get_FLIR.py#L174
Just make a very simple update that from
`if not calibP.calibrated:` to `if calibP.calibrated:`

And then the extractor will output a degree centigrade in geotiff files. That's a misunderstanding of the parameter in metadata of `"is calibrated"`, if we are looking for an output of Kelvin temperature we should delete the Kelvin parameter `273.15`.

ZongyangLi, 2017-09-07:
@max-zilla 
Please use the updated codes here: https://github.com/terraref/extractors-multispectral/blob/master/flir2tif/Get_FLIR.py

dlebauer, 2017-09-07:
The variable to be inserted into BETYdb is surface_temperature with units of K. See https://terraref.ncsa.illinois.edu/bety/variables/6000000198 or https://terraref.ncsa.illinois.edu/bety/api/beta/variables?key=9999999999999999999999999999999999999999&name=surface_temperature

dlebauer, 2017-09-07:
Questions about the extractor https://github.com/terraref/extractors-multispectral/blob/master/flir2tif/Get_FLIR.py

* [ ] there are many magic numbers that appear without comment, 
    * examples: gmin, gmax, H, T, D, E, HEIGHT_MAGIC_NUMBER 
* [ ] there are many functions that appear to belong in (and / or are redundant with) terrautils 
    * examples: createVrt, createMapTiles, generate_googlemaps, parse_metadata, get_bounding_box, get_bounding_box_with_formula, create_geotiff
    * are these all currently used in the pipeline?
* [ ] [Line 226](https://github.com/terraref/extractors-multispectral/blob/master/flir2tif/Get_FLIR.py#L226 ) says "TODO Something wonky w/ uint8s --> ending up w/ lots of gaps in data (white pixels)". Can this be fixed now? If not, please add to the quality statement and create a new github issue to track this.

max-zilla, 2017-09-08:
@dlebauer 

- merged your branch to output as units K, and updated bety trait name in meantemp extractor correspondingly.
- our extractor calls one method directly, rawData_to_temperature(), which calls:
  - get_calibrate_param(), pulling calibrations from metadata
  - flirRawToTemperature(), containing the following magic numbers/formula:
```
    H2O_K1 = 1.56
    H2O_K2 = 0.0694
    H2O_K3 = -0.000278
    H2O_K4 = 0.000000685
    
    H = 0.1
    T = 22.0
    D = 2.5
    E = 0.98

    K0 = 273.15
    
    im = rawData
        
    AmbTemp = T + K0
    AtmTemp = T + K0
        
    H2OInGperM2 = H*math.exp(H2O_K1 + H2O_K2*T + H2O_K3*math.pow(T, 2) + 
    H2O_K4*math.pow(T, 3))
    a1b1sqH2O = (a1+b1*math.sqrt(H2OInGperM2))
    a2b2sqH2O = (a2+b2*math.sqrt(H2OInGperM2))
    exp1 = math.exp(-math.sqrt(D/2)*a1b1sqH2O)
    exp2 = math.exp(-math.sqrt(D/2)*a2b2sqH2O)
        
    tao = X*exp1 + (1-X)*exp2
        
    obj_rad = im*E*tao
        
    theo_atm_rad = (R*J1/(math.exp(B/AtmTemp)-F)) + J0
    atm_rad = repmat((1-tao)*theo_atm_rad, 640, 480)
        
    theo_amb_refl_rad = (R*J1/(math.exp(B/AmbTemp)-F)) + J0
    amb_refl_rad = repmat((1-E)*tao*theo_amb_refl_rad, 640, 480)
        
    corr_pxl_val = obj_rad + atm_rad + amb_refl_rad
        
    pxl_temp = B/np.log(R/(corr_pxl_val-J0)*J1+F) - K0
```   
- for now, if @ZongyangLi can document this formula that is probably sufficient. we could put a short portion in README saying "the pixel values are converted to K using the following" or something.
- the other functions you mention are not used in our pipeline. our extractor code calls the terrautils versions of those things - those are likely old versions Zongyang used last year during development.
- line 226 is not reached anymore, we call terrautils version of create_geotiff. but because I used portions of this code as a basis, uint8 is still the format used: https://github.com/terraref/terrautils/blob/master/terrautils/formats.py#L75 we can first see if the white pixel issue still exists in our test outputs.

max-zilla, 2017-09-08:
@ZongyangLi OK, i have a flag in terrautils create_geotiff that lets use determine wtherh to use Float32 (FLIR) or Byte (RGB) so I think that comment can be removed.
```
if asfloat:
        output_raster = gdal.GetDriverByName('GTiff').Create(out_path, ncols, nrows, channels, gdal.GDT_Float32)
    else:
        output_raster = gdal.GetDriverByName('GTiff').Create(out_path, ncols, nrows, channels, gdal.GDT_Byte)
```

dlebauer, 2017-09-09:
@ZongyangLi and @max-zilla looking at the image in https://github.com/terraref/reference-data/issues/182#issuecomment-327835162 it appears there are some white pixels, consistent with the error mentioned in line 226. 

@ZongyangLi could you please update the code to remove obsolete lines / functions that have been migrated to terrautils?

ZongyangLi, 2017-09-09:
@dlebauer I think we had updated the codes relate to  the error mentioned in line 226. 

Actually lines from 226 to 237 was commented by line 225 and 238 `'''`, these lines are used to make a rgb full field map letting me have an easy way to review the map during my development. All the geotiff creations are using [line 218](https://github.com/terraref/extractors-multispectral/blob/master/flir2tif/Get_FLIR.py#L218).

@max-zilla shall we set up another short skype chat to discuss how to remove my initial codes on Github? I believe you had merge a lot of codes into terrautils.

max-zilla, 2017-09-11:
a new full field stitch has been created here:
```/sites/ua-mac/Level_1/fullfield/2017-08-01/```
https://terraref.ncsa.illinois.edu/clowder/datasets/59b681aa4f0c0b27bc24843f 

...created using images from 08/01 e.g.:
https://terraref.ncsa.illinois.edu/clowder/datasets/59b2ec6c4f0ca12ea0d7a162

...with corrected temperature conversion.

max-zilla, 2017-09-11:
In this stitch, the western half is consistently cooler than eastern:
[image]

...but only with a narrow K range (dark min = 305, light max = 315).
Using a broader color range from 60-120F (285-325K) the field is pretty regular with a small number of warmer plant blades:
[image]

pless, 2017-09-12:
Hi everyone,
 Something is clearly wrong with this stitched mosaic.  @ZongyangLi and I are trying to debug the problem.  We are exploring.
(a) is there something in the actual recent sensor data that suggests that one side of the image is consistently cooler than the other, or
(b) is there a bug in how we are converting the raw data to the temperature value that is causing us to have this trend, or
(c) is there a bug in the way that the stitching process proceeds that leads to this false trend.
We hope to have an answer asap.

max-zilla, 2017-09-12:
@pless thanks for the update. let me know if i can answer questions or point to to where code is being used.

ZongyangLi, 2017-09-12:
@max-zilla @dlebauer @pless 

After downloading the flir raw data, I happen to realize that the flir scan is also in a vertical directory, like the one in stereoTop scan. Would it be the reason why western area different from eastern area?

dlebauer, 2017-09-12:
@ZongyangLi the east-west temperature discrepancy occurs within an image, so should be independent of the file organization.

dlebauer, 2017-09-12:
eg https://terraref.ncsa.illinois.edu/clowder/datasets/59b044014f0ca12ea0bab961
[image]

max-zilla, 2017-09-12:
@ZongyangLi we did confirm that the flirIr is a vertical scan on this date, just like stereoTop (only X increases in gantry coordinates). so potentially, could explain why the field gets warmer as we move to the right (day is getting hotter).

doesn't explain why the individual images have that pattern, however.

max-zilla, 2017-09-12:
@ZongyangLi @craig-willis and I spot-checked a few datasets from 2016 with our old extractor and didn't see this pattern, but we did see this pattern starting with the May 2017 datasets.

doesn't answer the question - maybe something changed with sensor over winter, or the updated extractor code from terrautils isn't properly converting temperature (even though I would expect a problem across the entire image and not just in the left side then)

ZongyangLi, 2017-09-12:
@dlebauer Thanks a lot for helping me get the point of the problem.

@max-zilla Thanks for the check, I also downloaded some example data from different days in 2017.

An initial conclusion here: camera in the right-top is impacted by some unknown factor.
Start from 2017-05-03 to 2017-05-13 flir data is unreasonable like this:
[image]
after that, from 2017-05-16, pictures are getting back, but in the right-top corner, images are getting blurred.
[image]
In 05-25, it seems to me that the left side of image is clearer than the right side.
[image]
Things are getting worse after that:
[image]
[image]
[image]
Finally in July, images becomes to the picture we seen before:
[image]
[image]

I was thinking is it possible a camera problem rather than the extractor problem at this point.

smarshall-bmr, 2017-09-13:
This is pretty weird, I'm pulling data to check this. It almost looks like the autofocus is being confused or something got inside the camera housing.

max-zilla, 2017-09-13:
@ZongyangLi nice detective work.

smarshall-bmr, 2017-09-13:
https://drive.google.com/open?id=1MHtNlCjWWaD_mXx1Zw83lXmh5Jh8eZL6Ug
The outside of the FLIR has been cleaned pretty regularly. I'm checking with folks to see if I'm allowed to open the case to check to see if anything has gotten inside.

smarshall-bmr, 2017-09-15:
I tried running a whole bunch of tests today trying to pin down why the FLIR is making such terrible images and finally went with the nuclear option and opened the weatherproof case. Turns out the weatherproof case isn't Arizona-proof!
[image]

smarshall-bmr, 2017-09-15:
I looked back through the field book. There were sprinkler irrigations on 5/1 and 5/4. I'm thinking the inside of the case filled with some water causing the completely unreasonable data which I was working on troubleshooting when the FLIR suddenly just started working again. Because it was right at the beginning of the season after a bunch of power cycles I thought it was caused by the FLIR being too hot from one of the cooling circuits being turned off and I didn't look into it further.

Good news/bad news is that the problem should be pretty consistent over the whole season.

dlebauer, 2017-09-15:
We need to mark all FLIR data as having a known error since that date, and exclude it from the pipeline.

dlebauer, 2017-09-15:
PS thanks for tracking this down everyone. Let's discuss next steps with the team Tuesday and Wednesday.

NewcombMaria, 2017-09-15:
@dlebauer and @smarshall-bmr I'm not understanding fully when the problem with FLIR data began.  A full-field stitched Flir image was assembled by @max-zilla and @ZongyangLi using data collected the week of May 22.  My understanding from the glimpses in closed Github issue #327 ('Generate full field flirIR image from week of May 22') was that the images looked great.  We went into the Stomatal Regulation Field Campaign with some confidence that the full-field FlirIR suggested high quality data.  

[image]

@smarshall-bmr we were careful during the early season sprinkler runs (which ended May 4) to raise the camera box and keep it North of the sprinklers until either you or I ran it South so it would only intersect briefly with the sprinklers.  The only exception was a glitch in communication the night of April 28 that was documented in Slack.
 (https://terra-ref.slack.com/archives/C2MF45G2D/p1493403571162894)  
The camera box was checked after that night and all looked ok.  

I agree with @dlebauer comment to discuss the topic with the team Tue and Wed.  The thermal Flir data is critical to 2 of 3 field campaigns (stomatal regulation and stress effects on photosynthesis).  I'm hopeful from the data view the week of May 22 that some data may be good quality.

dlebauer, 2017-09-15:
Stuart if you haven't cleaned the camera yet I think a high priority will be to take some ground truth images to see if we can correct this. From Zongyang's images above it does seem the images were inconsistent. In the worst case the water  sloshed around as it evaporated over a the season. Are there any other potential sources of water? The AC perhaps? The sand and salt may be a separate source. Are there other instruments that need to be checked?

dlebauer, 2017-09-15:
Maria- we may be able to get useful data, but not in time for the November release.

dlebauer, 2017-09-15:
Maria in the image you posted in https://github.com/terraref/reference-data/issues/182#issuecomment-329789752 the leaves on the left appear cooler than the rest, which is consistent with what we have observed. The temperature scale may have made this less obvious when (wet?) soil was also in the scene.

NewcombMaria, 2017-09-15:
Thanks David.  Yes good suggestion to do what we can to allow ways to correct for this down the road, like ground-truth images etc.  Also looking into image overlap.

smarshall-bmr, 2017-09-15:
@dlebauer When we go to try and correct this should we take a set of images in standardized conditions, then clean out the camera and take a second set in the same conditions so we can more precisely determine what changed?

@NewcombMaria If memory serves correctly, wasn't there a pretty violent storm around this time? That would seem more likely then the sprinkler scenario.

Right now I'm thinking the cause for the inconsistencies is temperature related given that the housing lens is supposed to be IR transparent but the temperature of the salt on the lens is probably adding to the signal because the camera is focusing through it.

dlebauer, 2017-09-15:
@smarshall-bmr that sounds like a good plan but I'll defer to experts - Robert or folks from lemnatec & FLIR.

max-zilla, 2017-09-19:
https://terraref.ncsa.illinois.edu/clowder/datasets/59ba90ec4f0c0b27bc3cf7db

I generated full-field for FLIR on 05/29 so we have a day that has both FLIR and RGB stitches available.

There are also 05/29 Thermal IR tiles available.
https://terraref.ncsa.illinois.edu/clowder/collection/59c019e34f0c0b27bc3e9db4
e.g.
https://terraref.ncsa.illinois.edu/clowder/datasets/59c06ef24f0c0b27bc498515

max-zilla, 2017-09-21:
Generate a 2016 full field sample as well.

dlebauer, 2017-10-12:
Every through May 1, 2017 is OK; earlier data can be run. 

* Still need review of samples.
* mark data from May 1 2017 as bad. Process through geotiff but no further

closed 2018-06-07 by craig-willis