terraref/reference-data #184: Review Soil Mask extractor and output

dlebauer, 2017-09-08:
### Description 

Review issue for the soil mask extractor and output 

### Completion Criteria 

* [ ] Code review complete (includes extractor, any functions in terrautils, etc)
* [ ] Sample data output verified 
* [ ] Sample metadata verified
* [ ] Changes for 2017 release implemented
* [ ] Issue(s) for changes postponed until 2018 created and listed in README
* [ ] Extractor tagged/versioned
* [ ] Data from earlier versions of extractor purged
* [ ] Extractor deployed
* [ ] Final output verified/reviewed

dlebauer, 2017-09-08:
@Paheding could you please link to some sample output?

Paheding, 2017-09-13:
@dlebauer  An example output can be found here: 
/ua-mac/Level_1/soil_removal/VNIR/2017-05-08/2017-05-08__10-30-53-012/
[image]

I have incorporated the meta data into the second version of this extractor in the form of a **.netCDF** file, but I have not seen any output in the Globus. I will verify them once the outputs are available.

max-zilla, 2017-09-13:
the hyperspectral extractor must be run to generate this output, and i havent run that in a while since it seems like it's still being worked on.

Paheding, 2017-09-13:
@max-zilla  Thank you for letting me know this.  I will check it back later.