terraref/reference-data #186: Review canopy cover extractor & output

max-zilla, 2017-09-18:
Review issue for the canopy cover extractor and output using RGB full field input.  

Completion criteria:
* [ ] Code review complete (includes extractor, any functions in terrautils, etc)
* [ ] Sample data output verified 
* [ ] Sample metadata verified
* [ ] Changes for 2017 release implemented
* [ ] Issue(s) for changes postponed until 2018 created 
* [ ] Extractor tagged/versioned
* [ ] Extractor deployed
* [ ] Final output verified/reviewed

max-zilla, 2017-09-18:
The following RGB full-field image (05/29) was run through the extractor:
https://terraref.ncsa.illinois.edu/clowder/files/59bb28044f0c0b27bc3d5421

The datapoints were created in per-plot streams. You can find a datapoint by:
- going to list of plots
https://terraref.ncsa.illinois.edu/clowder/geostreams/sensors

- picking a plot ("sensor" in geostream) and click Edit
https://terraref.ncsa.illinois.edu/clowder/geostreams/sensors/586

- under the map, see the "STREAM ID: 3360" link to the right of "Canopy Cover" entry. the stream Id will change depending on which plot you choose. note the ID number

- Use the /api/geostreams/datapoints?stream_id=STREAMID endpoint to get datapoints for that stream:
https://terraref.ncsa.illinois.edu/clowder/api/geostreams/datapoints?stream_id=3360
(no credentials necessary)

There is currently one canopy cover entry for all the plots we have coverage of in full field mosaic. We will need to check the number, in the linked example the datapoint properties include source image and value:
```
properties: {
source: "https://terraref.ncsa.illinois.edu/clowder/files/59bb28044f0c0b27bc3d5421",
canopy_cover: 0.4618905450310325
}
...
stream_id: "3360",
sensor_id: "586",
sensor_name: "MAC Field Scanner Season 4 Range 25 Column 14"
```

Here, Range 25 Column 14 is reported as having 46.2% green pixel coverage on this date.

To check this, I loaded the _thumb.tif image from that dataset in QGIS and created a plot.js file with the GeoJSON from the sensors link:
https://terraref.ncsa.illinois.edu/clowder/geostreams/sensors/586
...you can then drag and drop the plot.js file into QGIS and see where it falls in the field:
[image]
...so 48% at least doesn't seem outrageous in either direction here.

dlebauer, 2017-09-18:
Looks good so far. Does this issue cover insertion into BETYdb?

max-zilla, 2017-09-18:
@dlebauer yes, this should've pushed 864 traits to BETYdb.

I put a link to the variable in the metadata:
https://terraref.ncsa.illinois.edu/bety/api/beta/variables?name=canopy_cover

...but I can't remember the right query to see the actual trait observations. They should be in there though, added today.

dlebauer, 2017-09-18:
https://terraref.ncsa.illinois.edu/bety/api/beta/search?trait=canopy_cover&raw_date=~2017-05-29&key=YOURSECRETKEY&limit=864

Here is a sample output: https://gist.github.com/dlebauer/2d7d4f8fe70ab1f8dcc08a5bebfe71a6

dlebauer, 2017-09-20:
I've purged data from BETYdb:

```sql
delete from traits where variable_id = (select id from variables where name = 'canopy_cover')
```

ZongyangLi, 2017-10-02:
**Canopy cover extractor statement**

This extractor processes binary stereo images and generates plot-level percentage canopy cover traits for BETYdb. The core idea for this extractor is a plant-soil segmentation. We apply a threshold to differentiate plant and soil, and do a smoothing after binary processing. At last it returns a plant area ratio within the bounding box.

We believe the tested threshold works well in a normal illumination. Below are some examples of segmentation:
[image]
[image]
[image]

At the same time, there are some limitation with the current threshold.
1. Image captured in a low illumination.
[image]

2. Image captured in a very high illumination.
[image]

3. In late season, panicle is covering a lot in the image, and leaves is getting yellow.
[image]

4. Sometimes sensor problem.
[image]

ZongyangLi, 2017-10-02:
@max-zilla 
In some lines I see a function name **CanopyCoverHeight**
like https://github.com/terraref/extractors-stereo-rgb/blob/master/canopycover/terra_canopycover.py#L25
and https://github.com/terraref/extractors-stereo-rgb/blob/master/canopycover/terra_canopycover.py#L27

Shall we modify these to avoid confusion？

max-zilla, 2017-10-02:
@ZongyangLi good idea, that was old name before I understood. will rename to CanopyCoverPercentage.

dlebauer, 2017-10-02:
@ZongyangLi I moved the quality statement to the README https://github.com/terraref/extractors-stereo-rgb/pull/17 - please flush it out there. Thanks!

remotesensinglab, 2017-10-03:
We have just developed an algorithm to enhance illumination condition of RGB images. The paper is in review right now. We can test on these datasets to address soil issues under shadow. I tagged Patrick on this issue. We will update soon.

dlebauer, 2017-10-03:
Wasit - this extractor is ready to go, had been tested, checked etc. Please create a new pull request or issue describing this feature and we can put it in the pipeline for the next release.

Paheding, 2017-10-03:
@dlebauer Yes, we can put it in the next release. As a reference, here is a preliminary result of the algorithm using the image shown above. 

**Left: Original image                  ----                    Right: Enhanced image**

[image]

dlebauer, 2018-03-08:
@ZongyangLi can you please review the output data? If you are logged in to BETYdb at https://terraref.ncsa.illinois.edu/bety you can download it from https://terraref.ncsa.illinois.edu/bety/search.csv?search=canopy+cover&include_unchecked=true

dlebauer, 2018-03-15:
needed to update output from fraction in range [0,1] to percent [0,100] https://github.com/terraref/reference-data/issues/234

dlebauer, 2019-01-31:
No longer relevant owing to new implementation. 

But I have marked all data generated by the old version of the extractor as 'checked=-1' i.e. 'checked and found to be in error' because it is clear that the algorithm got confused:

[image]

```sql
update traits set checked = -1 
   where  method_id in 
       (select id from methods where name = 'Canopy Cover Estimation from Field Scanner RGB images') 
   and site_id in 
      (select id from sites where sitename not like '%Season 6%');
-- UPDATE 82341
```

closed 2019-01-31 by dlebauer