terraref/reference-data #187: Review sensorposition extractor and output

max-zilla, 2017-09-19:
Review issue for the sensorposition extractor and output using Lemnatec metadata input.  

Completion criteria:
* [ ] Code review complete (includes extractor, any functions in terrautils, etc)
* [ ] Sample data output verified 
* [ ] Sample metadata verified
* [ ] Changes for 2017 release implemented
* [ ] Issue(s) for changes postponed until 2018 created 
* [ ] Extractor tagged/versioned
* [ ] Extractor deployed
* [ ] Final output verified/reviewed

max-zilla, 2017-09-19:
The Thermal IR 05/29 collection was run through this extractor:
https://terraref.ncsa.illinois.edu/clowder/collection/59c019e34f0c0b27bc3e9db4

You can select a plot from the list:
https://terraref.ncsa.illinois.edu/clowder/geostreams/sensors

e.g.:
https://terraref.ncsa.illinois.edu/clowder/geostreams/sensors/959

...and see plots are getting a "Thermal IR GeoTIFF Datasets" stream which will report all datasets from that sensor/data product for that plot. Raw data and Level_1 products alike will be registered this way.

The STREAM_ID to the right of the stream name can be used to get datapoints from geostreams API, e.g. in the linked example it is 4114:
https://terraref.ncsa.illinois.edu/clowder/api/geostreams/datapoints?stream_id=4114

max-zilla, 2017-09-19:
Looks like we need to fix the centroid calculation code, as it is currently assigning datasets to plots based on upper-right position and not centroid:
[image]
(green dot is where centroid is represented for https://terraref.ncsa.illinois.edu/clowder/datasets/59c067fc4f0c0b27bc489d6e)

@craig-willis i will take a look, this should be a quick fix.

The good news is that it's assigning things to the plots properly, it's just using the wrong point as a basis :)

max-zilla, 2017-09-19:
Fixed in this PR:
https://github.com/terraref/terrautils/pull/23

[image]

max-zilla, 2017-09-28:
@jterstriep an example of how to query the sensorposition outputs to get a list of dataset URLs that belong to this plot:
https://terraref.ncsa.illinois.edu/clowder/api/geostreams/datapoints?sensor_id=959&attributes=dataset_name&attributes=source_dataset

here, 959 is the plot ID (you can see by selecting a plot here: https://terraref.ncsa.illinois.edu/clowder/geostreams/sensors and looking at the URL) and it retrieves the "source_dataset" attribute only. Two results:

```
{
    id: 41085688,
    created: "2017-09-19T15:12:46Z",
    start_time: "2017-05-29T18:36:19Z",
    end_time: "2017-05-29T18:36:19Z",
    type: "Feature",
    geometry: {...},
    stream_id: "4114",
    sensor_id: "959",
    sensor_name: "MAC Field Scanner Season 4 Range 21 Column 4",
    properties: {
        dataset_name: "Thermal IR GeoTIFFs - 2017-05-29__11-36-19-149",
        source_dataset: "https://terraref.ncsa.illinois.edu/clowder/datasets/59c067fc4f0c0b27bc489d6e"
    }
},
{
    id: 41087949,
    created: "2017-09-19T17:01:06Z",
    start_time: "2017-05-29T18:36:20Z",
    end_time: "2017-05-29T18:36:20Z",
    type: "Feature",
    geometry: {...},
    stream_id: "4114",
    sensor_id: "959",
    sensor_name: "MAC Field Scanner Season 4 Range 21 Column 4",
    properties: {
        dataset_name: "Thermal IR GeoTIFFs - 2017-05-29__11-36-20-651",
        source_dataset: "https://terraref.ncsa.illinois.edu/clowder/datasets/59c04ecd4f0c0b27bc4550f9"
    }
},
...
```

This is determined at a "dataset" level. We could include ROGER paths of the files in each dataset here as well, although that might be overkill? That is what David had specified, so I should probably add that to sensorposition extractor. We could also include the link that lets you download each dataset as a .zip file...

craig-willis, 2018-06-07:
Stale issue -- closing. Open a new issue if work remains.

closed 2018-06-07 by craig-willis