terraref/reference-data #191: Review IR meantemp extractor & output

max-zilla, 2017-09-21:
Review issue for the meantemp extractor and output using IR full field input.  

Completion criteria:
* [ ] Code review complete (includes extractor, any functions in terrautils, etc)
* [ ] Sample data output verified 
   * [ ] Geostreams
   * [ ] BETYdb 
* [ ] Sample metadata verified
* [ ] Changes for 2017 release implemented
* [ ] Issue(s) for changes postponed until 2018 created 
* [ ] Extractor tagged/versioned
* [ ] Extractor deployed
* [ ] Final output verified/reviewed

max-zilla, 2017-09-21:
The following IR full-field image (05/29) was run through the extractor:
https://terraref.ncsa.illinois.edu/clowder/files/59c11b2a4f0c0b27bc4a59f9

The datapoints were created in per-plot streams. You can find a datapoint by:
- going to list of plots
https://terraref.ncsa.illinois.edu/clowder/geostreams/sensors

- picking a plot ("sensor" in geostream) and click Edit
https://terraref.ncsa.illinois.edu/clowder/geostreams/sensors/586
(you can also do this from the map view but that is tricky until we allow further zooming)

- under the map, see the "STREAM ID: 4865" link to the right of "IR Surface Temperature" entry. the stream Id will change depending on which plot you choose. note the ID number

- Use the /api/geostreams/datapoints?stream_id=STREAMID endpoint to get datapoints for that stream:
https://terraref.ncsa.illinois.edu/clowder/api/geostreams/datapoints?stream_id=4865
(no credentials necessary)

There is currently one canopy cover entry for all the plots we have coverage of in full field mosaic. We will need to check the number, in the linked example the datapoint properties include source image and value:
```
properties: {
source: "https://terraref.ncsa.illinois.edu/clowder/files/59c11b2a4f0c0b27bc4a59f9",
surface_temperature: "311.933"
}
...
stream_id: "4865",
sensor_id: "496",
sensor_name: "MAC Field Scanner Season 4 Range 30 Column 9"
```

dlebauer, 2017-09-28:
we are going to exclude data after May 2017. But can verify that mean T is being returned properly.

ZongyangLi, 2017-09-28:
@max-zilla As I could remember, I didn't write meantemp before, could you please point me out where's the meantemp part of script? I will take a look into it first.

max-zilla, 2017-09-28:
@ZongyangLi I actually built meantemp off of your flir2tif (GetFlir) script.
 
https://github.com/terraref/extractors-multispectral/blob/master/flir2tif/terra_flir2tif.py#L124

This is your code to convert raw_data_to_temperature (Which is a separate review issue).

https://github.com/terraref/extractors-multispectral/blob/master/meantemp/terra_meantemp.py

The meantemp script just opens the full field stitch of the FLIR files, clips them to the plot and takes the mean of numpy pixel array:
```
(pxarray, geotrans) = clip_raster(resource['local_paths'][0], bounds)
            mean_tc = numpy.mean(pxarray)

            # Create BETY-ready CSV
            (fields, traits) = get_traits_table()
            traits['surface_temperature'] = str(mean_tc)
            traits['site'] = plotname
            traits['local_datetime'] = timestamp+"T12-00-00-000"
            trait_list = generate_traits_list(traits)
            generate_csv(tmp_csv, fields, trait_list)

            # submit CSV to BETY
            submit_traits(tmp_csv, self.bety_key)
```

...so I guess if the flir2tif extractor is converting temperatures properly, there isn't much else to review here. I was thinking you had written the old FLIR converter to send this to BETYdb also but I think I was wrong, sorry.

craig-willis, 2018-06-13:
Stale issue. Will start new review process for 2018 in separate issue.

closed 2018-06-13 by craig-willis